espressif/esp-aws-iot #1: aws iot  aws_iot_shadow_register_delta api 问题 (IDFGH-994)

myzhang001, 2019-04-18:
在使用 aws  iot  sdk 的时候
调用 aws_iot_shadow_register_delta（）api 时候注册的字段超过9个 后
然后调用 aws_iot_shadow_update 就会导致复位

mahavirj, 2019-04-18:
@myzhang001 Could you please help to share entire application/test code with `sdkconfig`?

myzhang001, 2019-04-18:
`/*
this is   sdkconfig
 *
 * Automatically generated file; DO NOT EDIT.
 * Espressif IoT Development Framework Configuration
 *
 */
#define CONFIG_ESP32_PHY_MAX_TX_POWER 20
#define CONFIG_TRACEMEM_RESERVE_DRAM 0x0
#define CONFIG_FREERTOS_MAX_TASK_NAME_LEN 16
#define CONFIG_MQTT_TRANSPORT_SSL 1
#define CONFIG_MB_SERIAL_TASK_PRIO 10
#define CONFIG_MQTT_PROTOCOL_311 1
#define CONFIG_TCP_RECVMBOX_SIZE 6
#define CONFIG_FATFS_CODEPAGE_437 1
#define CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_ACKS 15
#define CONFIG_TCP_WND_DEFAULT 5744
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000
#define CONFIG_SPIFFS_USE_MAGIC_LENGTH 1
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_IPC_TASK_STACK_SIZE 1024
#define CONFIG_FATFS_PER_FILE_CACHE 1
#define CONFIG_ESPTOOLPY_FLASHFREQ "80m"
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_THING_NAME 30
#define CONFIG_MBEDTLS_KEY_EXCHANGE_RSA 1
#define CONFIG_UDP_RECVMBOX_SIZE 6
#define CONFIG_FREERTOS_QUEUE_REGISTRY_SIZE 0
#define CONFIG_MBEDTLS_AES_C 1
#define CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED 1
#define CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN 752
#define CONFIG_MBEDTLS_GCM_C 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_HEAP_POISONING_DISABLED 1
#define CONFIG_SPIFFS_CACHE_WR 1
#define CONFIG_BROWNOUT_DET_LVL_SEL_0 1
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER 1
#define CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE 1
#define CONFIG_SPIFFS_CACHE 1
#define CONFIG_INT_WDT 1
#define CONFIG_MBEDTLS_SSL_PROTO_TLS1 1
#define CONFIG_ESP_GRATUITOUS_ARP 1
#define CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_UNIQUE_CLIENT_ID_BYTES 80
#define CONFIG_MBEDTLS_ECDSA_C 1
#define CONFIG_LOG_BOOTLOADER_LEVEL_INFO 1
#define CONFIG_HTTPD_MAX_REQ_HDR_LEN 512
#define CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE 0
#define CONFIG_AWS_IOT_MQTT_PORT 8883
#define CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS 1
#define CONFIG_MBEDTLS_ECDH_C 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE 1
#define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
#define CONFIG_AWS_IOT_MQTT_MAX_RECONNECT_WAIT_INTERVAL 128000
#define CONFIG_MBEDTLS_SSL_ALPN 1
#define CONFIG_MBEDTLS_PEM_WRITE_C 1
#define CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN 4096
#define CONFIG_LOG_DEFAULT_LEVEL_INFO 1
#define CONFIG_BT_RESERVE_DRAM 0x0
#define CONFIG_APP_COMPILE_TIME_DATE 1
#define CONFIG_FATFS_FS_LOCK 0
#define CONFIG_IP_LOST_TIMER_INTERVAL 120
#define CONFIG_SPIFFS_META_LENGTH 4
#define CONFIG_ESP32_PANIC_PRINT_REBOOT 1
#define CONFIG_MB_CONTROLLER_NOTIFY_QUEUE_SIZE 20
#define CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED 1
#define CONFIG_AWS_IOT_MQTT_RX_BUF_LEN 2048
#define CONFIG_MB_SERIAL_BUF_SIZE 256
#define CONFIG_CONSOLE_UART_BAUDRATE 115200
#define CONFIG_LWIP_MAX_SOCKETS 10
#define CONFIG_LWIP_NETIF_LOOPBACK 1
[... 196 lines cut ...]
#define CONFIG_SPI_SLAVE_ISR_IN_IRAM 1
#define CONFIG_WIFI_PASSWORD "12345678"
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_FATFS_MAX_LFN 255
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 1
#define CONFIG_BOOTLOADER_WDT_ENABLE 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN 16384
#define CONFIG_MB_TIMER_GROUP 0
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_MQTT_TRANSPORT_WEBSOCKET_SECURE 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 3072
#define CONFIG_MB_TIMER_PORT_ENABLED 1
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "/dev/ttyUSB0"
#define CONFIG_UNITY_ENABLE_DOUBLE 1
#define CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN 1
#define CONFIG_AWS_EXAMPLE_THING_NAME "PN777777777777777"
#define CONFIG_FATFS_API_ENCODING_ANSI_OEM 1
`

myzhang001, 2019-04-18:
`#include "user_aws_handle.h"

#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <unistd.h>
#include <limits.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"

#include "nvs.h"
#include "nvs_flash.h"

#include "aws_iot_config.h"
#include "aws_iot_log.h"
#include "aws_iot_version.h"
#include "aws_iot_mqtt_client_interface.h"
#include "aws_iot_shadow_interface.h"

#include "user_smart_config.h"
#include "system_parameter.h"

/*!
 * The goal of this sample application is to demonstrate the capabilities of shadow.
 * This device(say Connected Window) will open the window of a room based on temperature
 * It can report to the Shadow the following parameters:
 *  1. temperature of the room (double)
 *  2. status of the window (open or close)
 * It can act on commands from the cloud. In this case it will open or close the window based on the json object "windowOpen" data[open/close]
 *
 * The two variables from a device's perspective are double temperature and bool windowOpen
 * The device needs to act on only on windowOpen variable, so we will create a primitiveJson_t object with callback
 The Json Document in the cloud will be
 {
 "reported": {
 "temperature": 0,
 "windowOpen": false
 },
 "desired": {
 "windowOpen": false
 }
 }
 */

#define ROOMTEMPERATURE_UPPERLIMIT 32.0f
#define ROOMTEMPERATURE_LOWERLIMIT 25.0f
#define STARTING_ROOMTEMPERATURE ROOMTEMPERATURE_LOWERLIMIT

#define MAX_LENGTH_OF_UPDATE_JSON_BUFFER 200*5

static const char *TAG = "shadow_user";

/* FreeRTOS event group to signal when we are connected & ready to make a request */
extern EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
extern const int CONNECTED_BIT;
extern const int SERVER_BIT;

[... 1451 lines cut ...]
									}
								}
							}

							//ESP_LOGI(TAG, "*****************************************************************************************");
							ESP_LOGI(TAG, "Stack remaining for task '%s' is %d bytes", pcTaskGetTaskName(NULL), uxTaskGetStackHighWaterMark(NULL));
						}
					}	
					vTaskDelay(1000 / portTICK_RATE_MS);
					break;

				case 2:					  //变化上传状态数据

					break;
				default:
					break;
			}
			#endif
			

		 }	
	}
}

`

mahavirj, 2019-04-19:
@myzhang001 I am having trouble understanding your problem. Probably you are using `thing_shadow` example, can you please provide your changes in `zip` format so that it is easier to understand?

Also need help on some other questions:
 - Is problem reproducible with default `thing_shadow` example in IDF?
 - Which IDF version you are using?

closed 2019-04-30 by myzhang001