espressif/esp-aws-iot #2: Error compiling subscribe publish example - timer_platform.h no such file

borch84, 2019-04-23:
Hi, today I git cloned the repository:

`$ git clone https://github.com/espressif/esp-aws-iot.git`

I changed directory to examples/subscribe_publish and executed make command however I got this:

```
/home/borch/Documents/Robotica/esp/esp-aws-iot/port/include/**timer_platform.h:25:29: fatal error: timer_interface.h**: No such file or directory
compilation terminated.
/home/borch/Documents/Robotica/esp/esp-idf-master/esp-idf/make/component_wrapper.mk:289: recipe for target 'port/network_mbedtls_wrapper.o' failed
make[1]: *** [port/network_mbedtls_wrapper.o] Error 1
/home/borch/Documents/Robotica/esp/esp-idf-master/esp-idf/make/project.mk:582: recipe for target 'component-esp-aws-iot-build' failed
make: *** [component-esp-aws-iot-build] Error 2

```
The last commit id is:
```
git rev-parse HEAD
7a5d204da1bdf7a3d3d8ee7f7287ce87acf29048
```

The version of the IDF I am using is located at:

```
$ printenv IDF_PATH
/home/borch/Documents/Robotica/esp/esp-idf-master/esp-idf

$cd /home/borch/Documents/Robotica/esp/esp-idf-master/esp-idf
$ git rev-parse HEAD
a20d02b7f196c407bc9f39b781e31a0a4f665968 <== This is the last commit id of the master branch. 
```

mahavirj, 2019-04-24:
@borch84 Can you try cloning repository recursively as mentioned in https://github.com/espressif/esp-aws-iot/blob/master/README.md?

wltn928, 2019-04-25:
@borch84 
when download with "git clone" ,
you must use "--recursive" option 
it can also download "aws-iot-device-sdk-embedded-C" (included header file)

borch84, 2019-04-25:
Hi @mahavirj, @wltn928 I was able to compile the example successfully. Thank you. I did not include the --recursive option. Regards

closed 2019-04-25 by borch84