espressif/esp-aws-iot #3: Change json data in thing_shadow_sample.c

wltn928, 2019-04-29:
Hi, 

About change json data(document) in thing_shadow_sample.c

[image]

How to fix it?

I want to put the my data in the "char JsonDocumentBuffer" 
(I do not know when the data is initialized.)

Thank you for reading.
Yours faithfully.

mahavirj, 2019-04-30:
@wltn928 You may customize JSON data by modifying application, please see below where JSON document gets created, 
https://github.com/espressif/esp-aws-iot/blob/7a5d204da1bdf7a3d3d8ee7f7287ce87acf29048/examples/thing_shadow/main/thing_shadow_sample.c#L303

wltn928, 2019-05-02:
@mahavirj 
Thanks to reply..

ex) 
char JsonDocumentBuffer[MAX_LENGTH_OF_UPDATE_JSON_BUFFER]="{\"state\": { \"desired\": {\"welcome\": \"test\" } } }";

Do you mean to do this??
I'm really sorry.. but I have no confidence using JSON documents..

mahavirj, 2019-05-07:
@wltn928 I meant that you can use APIs provided by AWS IoT SDK itself,  e.g. https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/299183238ffe7a3e6a5ca0af9db19c10eaca62cf/include/aws_iot_shadow_json_data.h#L117 allows you to add desired section of JSON document using `key-pair` value from https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/299183238ffe7a3e6a5ca0af9db19c10eaca62cf/include/aws_iot_shadow_json_data.h#L63

wltn928, 2019-05-08:
@mahavirj 
Thanks for your help!! Thank you!!

closed 2019-05-08 by wltn928