espressif/esp-aws-iot #4: undefined reference to "FUNCTION"

wltn928, 2019-05-09:
Hi,

I am trying to integrate thing_shadow and ble_prov(in esp-idf/example)
( set wifi with ble_prov -> shadow_get )

Add
`#include <esp_bt.h>`
and make command there are problems like below
[image]

It looks like it doesn't refer to a function due to a library problem.
Can I link library??

Thanks to read!! 
best regards.

mahavirj, 2019-05-09:
@wltn928 Have you enabled `Bluetooth` from configuration option (`CONFIG_BT_ENABLED=y` have this line in your sdkconfig)?

wltn928, 2019-05-16:
@mahavirj 
I honestly respect you.
Thanks for help!!

closed 2019-05-16 by wltn928