espressif/esp-aws-iot #7: Publish/Subscribe: aws_iot failed! mbedtls_ssl_handshake returned -0x10

hanhdt, 2019-06-07:
Hi,
I tried to connect AWS IoT for testing but got this issue, could you have any suggestions?
Here is logs:
```
...
I (157526) mbedtls: ssl_tls.c:2770 message length: 877, out_left: 877
I (157536) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 877 (-0xfffffc93)
I (157546) mbedtls: ssl_tls.c:2803 <= flush output
I (157546) mbedtls: ssl_tls.c:3470 <= write record
I (157556) mbedtls: ssl_tls.c:3314 <= write handshake message
I (157566) mbedtls: ssl_tls.c:5427 <= write certificate
I (157566) mbedtls: ssl_cli.c:3405 client state: 8
I (157576) mbedtls: ssl_tls.c:2751 => flush output
I (157576) mbedtls: ssl_tls.c:2763 <= flush output
I (157586) mbedtls: ssl_cli.c:2851 => write client key exchange
W (158606) mbedtls: ssl_cli.c:2924 mbedtls_ecdh_calc_secret() returned -16 (-0x0010)
I (158606) mbedtls: ssl_tls.c:8031 <= handshake
E (158616) aws_iot: failed! mbedtls_ssl_handshake returned -0x10
I (158616) mbedtls: ssl_tls.c:8662 => write close notify
I (158626) mbedtls: ssl_tls.c:8678 <= write close notify
I (158636) mbedtls: ssl_tls.c:8866 => free
I (158636) mbedtls: ssl_tls.c:8931 <= free
```

Thank you.

shahpiyushv, 2019-06-07:
@hanhdt The error code 0x0010 indicates MBEDTLS_ERR_MPI_ALLOC_FAILED. Does you application code have many/big memory allocations?

hanhdt, 2019-06-08:
@shahpiyushv thanks for your quick response.
Actually I have used BLE and WiFi together and it might take a lot memory at that moments.
So, I will update to reduce memory allocation.

closed 2019-06-08 by hanhdt