espressif/esp-aws-iot #9: Thing Shadow example stops working after 5-10min (IDFGH-1604)

gearsmotion789, 2019-08-01:
I've ran the thing_shadow example from https://github.com/espressif/esp-aws-iot/tree/master/examples/thing_shadow succesfully. However, after 5-10min of inactivity (meaning that I don't update the device shadow), the device doesn't respond to anymore changes to the shadow. I don't see that it disconnected from AWS on the monitor - **so why is this happening and how can I fix this?**

I've modified the code slightly so that it only sends a "reported" msg to the device shadow every time there is a delta change. It doesn't update the shadow itself every 1s like the original code.

```
/*
 * Copyright 2010-2015 Amazon.com, Inc. or its affiliates. All Rights Reserved.
 * Additions Copyright 2016 Espressif Systems (Shanghai) PTE LTD
 *
 *
 * Licensed under the Apache License, Version 2.0 (the "License").
 * You may not use this file except in compliance with the License.
 * A copy of the License is located at
 *
 *  http://aws.amazon.com/apache2.0
 *
 * or in the "license" file accompanying this file. This file is distributed
 * on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either
 * express or implied. See the License for the specific language governing
 * permissions and limitations under the License.
 */
/**
 * @file thing_shadow_sample.c
 * @brief A simple connected window example demonstrating the use of Thing Shadow
 *
 * See example README for more details.
 */
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <unistd.h>
#include <limits.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"

#include "nvs.h"
#include "nvs_flash.h"

#include "aws_iot_config.h"
#include "aws_iot_log.h"
#include "aws_iot_version.h"
#include "aws_iot_mqtt_client_interface.h"
#include "aws_iot_shadow_interface.h"

/*!
 * The goal of this sample application is to demonstrate the capabilities of shadow.
 * This device(say Connected Window) will open the window of a room based on temperature
 * It can report to the Shadow the following parameters:
 *  1. temperature of the room (double)
 *  2. status of the window (open or close)
 * It can act on commands from the cloud. In this case it will open or close the window based on the json object "windowOpen" data[open/close]
 *
 * The two variables from a device's perspective are double temperature and bool windowOpen
 * The device needs to act on only on windowOpen variable, so we will create a primitiveJson_t object with callback
 The Json Document in the cloud will be
 {
 "reported": {
 "temperature": 0,
 "windowOpen": false
 },
 "desired": {
[... 312 lines cut ...]
            .ssid = EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
}

void app_main()
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    initialise_wifi();
    /* Temporarily pin task to core, due to FPU uncertainty */
    xTaskCreatePinnedToCore(&aws_iot_task, "aws_iot_task", 9216, NULL, 5, NULL, 1);
}

```

gearsmotion789, 2019-08-01:
@chegewara I tried doing that, but the callback was never triggered.
```
void disconnectCallback() {
    printf("\n\nHI I am disconnected\n\n");
}
-----------------
sp.disconnectHandler = disconnectCallback;
```

shahpiyushv, 2019-08-01:
@gearsmotion789 is your problem similar to this: https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/161?

gearsmotion789, 2019-08-01:
@shahpiyushv No. It doesn't seem to disconnect or reconnect according to the monitor.

gearsmotion789, 2019-08-02:
@shahpiyushv I believe that it is the same issue. I have changed keepAliveIntervalInSec from 600 to 65 in aws_iot_shadow.c and I no longer see this problem. **However, I wanted to ask is there any negatives to doing this (e.g more frequent polling of AWS IoT = more $$$)? Is there a better solution that doing this?**

shahpiyushv, 2019-08-02:
@gearsmotion789 there is no impact on cost as AWS pricing is based on MQTT messages and connection time, not the exchange of keep-alive messages. Meanwhile, can you check for a slightly larger value, say 300 seconds?

gearsmotion789, 2019-08-02:
@shahpiyushv Also, according to the link you gave, it says "The client disconnects at 1.5 times the keep alive value". How can I prevent the client from ever disconnecting?

gearsmotion789, 2019-08-04:
@shahpiyushv Thanks. So I wanted to understand why this problem happened. It seems the device resets the keepAliveIntervalInSec (so it doesn't disconnect) whenever AWS IoT receives a PUBLISH, SUBSCRIBE, PING, or PUBACK message from the client according to https://github.com/awslabs/aws-iot-device-sdk-cpp-v2/issues/10. When does this happen in my code & at what interval? - why did it fail when keepAliveIntervalInSec = 600?

shahpiyushv, 2019-10-16:
@gearsmotion789 , even I do not really understand why the issue occurs for 600 seconds and that too only on specific networks. If there is an issue with the network, which blocks the keep alive, it should have failed for any other interval as well.

azilberbaum, 2021-09-08:
Does anyone have any updates on this issue?