espressif/esp-aws-iot #10: subscribe publish example run time crash.

Monika121, 2019-08-14:
I am new to this environment and may be not able to understand the output log. This code is crashing i have given the output below,Please take a look, Thanks in advance.
 I am working with
1. ESP-SOLO 1 board which is single core .
2. IDF-Version v3.2.
3. I have configured all menuconfig configrations and re-checked the same.
4. warning in code are unused decleration of 3 variables 
   aws_root_ca_pem_end,certificate_pem_crt_end and private_pem_key_end

output is:
$make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/Chipwire/esp/esp-idf/requirements.txt ar                                                                                                                e satisfied.
MONITOR
--- idf_monitor on COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6196
load:0x40078000,len:10180
ho 0 tail 12 room 4
load:0x40080400,len:6640
entry 0x40080760
I (30) boot: ESP-IDF v3.2-dirty 2nd stage bootloader
I (30) boot: compile time 11:07:50
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18ff8 (102392) map
I (130) esp_image: segment 1: paddr=0x00029020 vaddr=0x3ffb0000 size=0x027f0 ( 10224) load
I (134) esp_image: segment 2: paddr=0x0002b818 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (137) esp_image: segment 3: paddr=0x0002bc20 vaddr=0x40080400 size=0x043f0 ( 17392) load
I (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x81838 (530488) map
0x400d0018: _stext at ??:?

I (340) esp_image: segment 5: paddr=0x000b1858 vaddr=0x400847f0 size=0x0c430 ( 50224) load
0x400847f0: ieee80211_output_process at ??:?

I (371) boot: Loaded app from partition at offset 0x10000
I (371) boot: Disabling RNG early entropy source...
I (371) cpu_start: Pro cpu up.
I (375) cpu_start: Single core mode
I (379) heap_init: Initializing. RAM available for dynamic allocation:
I (386) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (392) heap_init: At 3FFB8778 len 00027888 (158 KiB): DRAM
I (399) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (405) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (411) heap_init: At 40090C20 len 0000F3E0 (60 KiB): IRAM
I (417) cpu_start: Pro cpu start user code
I (100) cpu_start: Starting scheduler on PRO CPU.
I (123) wifi: wifi driver task: 3ffbf1b8, prio:23, stack:3584, core=0
I (123) wifi: wifi firmware version: 9415913
I (123) wifi: config NVS flash: enabled
I (123) wifi: config nano formating: disabled
I (123) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[... 86 lines cut ...]
I (153) wifi: Init management frame dynamic rx buffer num: 32
I (163) wifi: Init static rx buffer size: 1600
I (163) wifi: Init static rx buffer num: 10
I (173) wifi: Init dynamic rx buffer num: 32
I (173) subpub: Setting WiFi configuration SSID Redmi...
I (233) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0
I (233) wifi: mode : sta (24:0a:c4:85:ca:64)
C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/tasks.c:1069 (prvAddNewTaskToReadyList)- assert failed!
abort() was called at PC 0x4008b73a on core 0
0x4008b73a: prvAddNewTaskToReadyList at C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/tasks.c:3537

Backtrace: 0x4008e564:0x3ffb9a20 0x4008e771:0x3ffb9a40 0x4008b73a:0x3ffb9a60 0x4008b8a3:0x3ffb9a80 0x400d2db9:0x3ffb9ac0 0x400d0e84:0x3ffb9af0 0x4008c4e1:0x3ffb9b10
0x4008e564: invoke_abort at C:/msys32/home/Chipwire/esp/esp-idf/components/esp32/panic.c:707

0x4008e771: abort at C:/msys32/home/Chipwire/esp/esp-idf/components/esp32/panic.c:707

0x4008b73a: prvAddNewTaskToReadyList at C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/tasks.c:3537

0x4008b8a3: xTaskCreatePinnedToCore at C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/tasks.c:3537

0x400d2db9: app_main at C:/msys32/home/Chipwire/esp/esp-aws-iot-master/examples/subscribe_publish/main/subscribe_publish_sample.c:331

0x400d0e84: main_task at C:/msys32/home/Chipwire/esp/esp-idf/components/esp32/cpu_start.c:494

0x4008c4e1: vPortTaskWrapper at C:/msys32/home/Chipwire/esp/esp-idf/components/freertos/port.c:403

shahpiyushv, 2019-08-14:
@Monika121, can you try by changing the last parameter of xTaskCreatePinnedToCore()  from 1 to 
tskNO_AFFINITY at
https://github.com/espressif/esp-aws-iot/blob/master/examples/subscribe_publish/main/subscribe_publish_sample.c#L331?

Monika121, 2019-08-19:
Thank you for your reply @shahpiyushv , i tried the above but the output is still same.I have one doubt i am using esp SOLO-1 board which is of single core can that be the issue .

Monika121, 2019-08-19:
Thank you for your reply @shahpiyushv , i tried the above but the output is still same.

shahpiyushv, 2019-08-19:
@Monika121 , I had thought that tskNO_AFFINITY should work fine for the ESP-Solo module, but seems that you need to explicitly set the core number to 0. So, can you try replacing tskNO_AFFINITY with 0? If even that does not work, please replace the xTaskCreatePinnedToCore() function call with xTaskCreate().

Monika121, 2019-08-19:
Thank you @shahpiyushv ,Things are working fine with xTaskCreate().

closed 2019-08-19 by Monika121