espressif/esp-aws-iot #12: Not Able to include new header

Monika121, 2019-08-22:
I am new to this environment .I am planning to make project in which esp32 scans bluetooth beacons and send the rssi value to the aws cloud. So i am trying to take subcribe publish example as basecode and merging the gatt_client code in this code in which I am trying to include the header files.It is expected that all files of components under esp-idf can be built without any perticular chnage in makefile or component.mk. 
Please let me know where i am wrong ,all suggestions are welcome .
This is the error:

C:/msys32/home/monika/esp/esp-aws-iot-master/examples/subscribe_publish/main/subscribe_publish_sample.c:53:29: fatal error: esp_gap_ble_api.h: No such file or directory

ESP-SOLO 1 board which is single core ,IDF-Version v3.2.

shahpiyushv, 2019-08-22:
@Monika121 , Can you check if `CONFIG_BT_ENABLED=y` is set in your sdkconfig?

Monika121, 2019-08-22:
Thank you @shahpiyushv problem resloved. I was really confused and found this silly mistake.Thank you so much for your support .

closed 2019-08-22 by Monika121