espressif/esp-aws-iot #13: Update to v4.0 aws-iot-device-sdk-embedded-C?

WiktorBAbove, 2019-08-27:
Hi,
Will there be added support for:
https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta
?

copercini, 2019-09-19:
I'm also looking for this to use AWS OTA jobs

crippsJack, 2019-09-24:
Yes, any news when there will be an official port for v4 SDK to ESP-IDF as the current version is out of date compared to the ESP-IDF releases and not really up to production standards

crippsJack, 2019-10-04:
Now that the v4 SDK is no longer in beta, do we expect to see a port for this in the near future?

shahpiyushv, 2019-10-04:
@crippsJack , I tried giving it a shot but realised that there are quite some changes in the SDK and so, porting could take some time. I will check once again next week. There are indeed some useful features in v4!

crippsJack, 2019-10-04:
I have also begun porting but have ran into issues linking the esp mbedtls library port into the new version of SDK, since the mbedtls default port included in the SDK does not seem to work with esp-idf. Current problem lies with setting up the mbedtls mutex types to work in esp mbedtls port, just can’t get it to link correctly. Will update more on Monday if anybody is interested.

Thanks,
Jack

shahpiyushv, 2019-10-04:
@crippsJack , which ESP IDF version are you using?

crippsJack, 2019-10-04:
@shahpiyushv release/v4.0

crippsJack, 2019-10-07:
@shahpiyushv More specifically, I get undefined reference errors for the mutex threading alt functions, 
`mbedtls_threading_set_alt()` and `mbedtls_threading_free_alt()` at the linking stage - everything seems to compile ok.
I have `MBEDTLS_THREADING_C` (a prerequisite of the following) and  `MBEDTLS_THREADING_ALT` defined in the iot_config.h file of the SDK v4, though not sure if these definitions are 'global' and can be seen by the mbedtls esp component, will take a look at this.

`mbedtls_net_poll()` also gets an undefined reference error but this is because this function is not implemented in the ESP-IDF version of net_sockets.c, so this may be some more work required to get the port up and running.

crippsJack, 2019-10-07:
`mbedtls_threading_...` functions now appear to link correctly after defining `MBEDTLS_THREADING_C`
followed by `MBEDTLS_THREADING_ALT` in `esp_config.h` of mbedtls esp component.

WiktorBAbove, 2019-10-09:
@crippsJack  did you manage to get something working?

crippsJack, 2019-10-09:
@WiktorBAbove Everything now compiles, just working through runtime issues. An oversight was that the logging library in this version of the SDK is different to previous, and so will need proper porting so that the system can be debugged, information recorded etc. Will post updates in this issue

shahpiyushv, 2019-10-09:
@crippsJack , that is good to know!
If you can share the code, we can work on integrating it into this repo.

crippsJack, 2019-10-09:
@shahpiyushv I'm just finishing stuff off, sorting the logging library out then will share code - hopefully by the end of this week

crippsJack, 2019-10-14:
@shahpiyushv @mahavirj Ok, so compiling is fine but waiting on a semaphore always fails. Mutexes do seem to be able to be locked, as can be seen by the program log below.

This is the relevant bit of the stack trace for the crash:

```
I (2561) tcpip_adapter: sta ip: 10.10.10.146, mask: 255.255.255.0, gw: 10.10.10.1
I (2562) Sub-Pub: WiFi Connected, proceeding with AWS SDK Initialisation and MQTT Demo
I (2568) MAIN: Finished cert parsing
I (2573) CLOCK: Time(ms): 2054, Time buffer: 2054 SSB
[DEBUG][THREAD][] Entered semaphore create, init: 0, max: 65535
I (2583) CLOCK: Time(ms): 2065, Time buffer: 2065 SSB
[DEBUG][THREAD][] Created counting semaphore
I (2593) CLOCK: Time(ms): 2075, Time buffer: 2075 SSB
[DEBUG][THREAD][] Entered mutex create
I (2602) CLOCK: Time(ms): 2084, Time buffer: 2084 SSB
[DEBUG][THREAD][] Attempt recursive mutex create
I (2613) CLOCK: Time(ms): 2095, Time buffer: 2095 SSB
[DEBUG][THREAD][] Mutex create succesful
I (2622) CLOCK: Time(ms): 2104, Time buffer: 2104 SSB
[DEBUG][THREAD][] Entered semaphore create, init: 0, max: 65535
I (2633) CLOCK: Time(ms): 2115, Time buffer: 2115 SSB
[DEBUG][THREAD][] Created counting semaphore
I (2643) CLOCK: Time(ms): 2125, Time buffer: 2125 SSB
[DEBUG][THREAD][] Entered mutex lock
I (2652) CLOCK: Time(ms): 2134, Time buffer: 2134 SSB
[DEBUG][THREAD][] Locked a recursive mutex
I (2662) CLOCK: Time(ms): 2143, Time buffer: 2143 SSB
[DEBUG][THREAD][] Creating new thread.
I (2671) CLOCK: Time(ms): 2153, Time buffer: 2153 SSB
[DEBUG][THREAD][] Creating new thread.
I (2680) CLOCK: Time(ms): 2162, Time buffer: 2162 SSB
[DEBUG][THREAD][] Entered mutex unlock
I (2689) CLOCK: Time(ms): 2171, Time buffer: 2171 SSB
[DEBUG][THREAD][] Succesfully unlocked recursive mutex
I (2700) CLOCK: Time(ms): 2182, Time buffer: 2182 SSB
[DEBUG][THREAD][] Entered semaphore wait
/home/vlad/esp/esp-idf/components/freertos/queue.c:720 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008d790 on core 0
0x4008d790: xQueueGenericSend at /home/vlad/esp/esp-idf/components/freertos/queue.c:720 (discriminator 3)

ELF file SHA256: 856aec52d551e89dbb54ad0b0adf5cd57c36b7ac176489b45ae95b50546bf987

Backtrace: 0x4008abdd:0x3ffc89d0 0x4008af5d:0x3ffc89f0 0x4008d790:0x3ffc8a10 0x400de569:0x3ffc8a50 0x400d898b:0x3ffc8a80 0x400d905a:0x3ffc8aa0 0x400d8566:0x3ffc8ac0 0x400d7248:0x3ffc8af0 0x4008d149:0x3ffc8b60
0x4008abdd: invoke_abort at /home/vlad/esp/esp-idf/components/esp32/panic.c:155

0x4008af5d: abort at /home/vlad/esp/esp-idf/components/esp32/panic.c:172

0x4008d790: xQueueGenericSend at /home/vlad/esp/esp-idf/components/freertos/queue.c:720 (discriminator 3)

0x400de569: IotSemaphore_Wait at /home/vlad/git_folder/esp-aws-sdk-v4/AWS-SDK-v4_esp/platform/ports/esp32/source/iot_threads_template.c:364

0x400d898b: _createTaskPool at /home/vlad/git_folder/esp-aws-sdk-v4/AWS-SDK-v4_esp/lib/source/common/iot_taskpool.c:1103 (discriminator 3)

0x400d905a: IotTaskPool_CreateSystemTaskPool at /home/vlad/git_folder/esp-aws-sdk-v4/AWS-SDK-v4_esp/lib/source/common/iot_taskpool.c:281 (discriminator 2)

0x400d8566: IotSdk_Init at /home/vlad/git_folder/esp-aws-sdk-v4/AWS-SDK-v4_esp/lib/source/common/iot_init.c:103

0x400d7248: func at /home/vlad/git_folder/esp-aws-sdk-v4/examples/sub_pub/build/../main/sub_pub_example.c:947 (discriminator 9)

0x4008d149: vPortTaskWrapper at /home/vlad/esp/esp-idf/components/freertos/port.c:143
```
Relevant Code:

`typedef SemaphoreHandle_t IotSemaphore_t`

```
bool IotSemaphore_Create( IotSemaphore_t * pNewSemaphore,
                          uint32_t initialValue,
                          uint32_t maxValue )
{
    /* Implement this function as specified here:
     * https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/platform/platform_threads_function_semaphorecreate.html
     */
    IotLogDebug("Entered semaphore create, init: %d, max: %d", initialValue, maxValue);
    bool ret = false;

    pNewSemaphore = xSemaphoreCreateCounting(maxValue, initialValue);

    if (pNewSemaphore != NULL)
    {
        ret = true;
        IotLogDebug("Created counting semaphore");
    }
    else
    {
        ret = false;
        IotLogDebug("Failed to create counting semaphore");
    }

    return ret;
}
```

```
void IotSemaphore_Wait( IotSemaphore_t * pSemaphore )
{
    /* Implement this function as specified here:
     * https://docs.aws.amazon.com/freertos/latest/lib-ref/c- 
     * sdk/platform/platform_threads_function_semaphorewait.html
     */
    IotLogDebug("Entered semaphore wait");

    if (xSemaphoreTake(pSemaphore, portMAX_DELAY) == pdTRUE)
    {
        IotLogDebug("Semaphore waited on");
    }
    else
    {
        IotLogDebug("Failed wait on semaphore");
    }
}
```
If I set the wait time to zero the program doesn't crash here but the sem wait does still fail and causes a different crash later.

Any ideas? The porting process for this SDK does not seem worth the development effort so am considering moving back to the previous version of AWS SDK

WiktorBAbove, 2019-10-17:
Good job @crippsJack!
You think it would be possible for you to push the current work to a dev-branch?
So that other ppl can maybe continue your work?

crippsJack, 2019-10-17:
@WiktorBAbove Yes I will be able to, got some other things on at the moment, but will tidy up and push ASAP

jkent, 2019-10-18:
There is the official amazon port for FreeRTOS: https://aws.amazon.com/freertos/ -- I've checked it out and they're using static semaphores.  Its under the MIT license.

jkent, 2019-10-20:
I'm working on this here:  https://github.com/jkent/esp-aws-iot

I've not tried to use it yet, please report back if you try it!

Schuck84, 2020-05-19:
Hi,
Is this repo compatible with ESP-IDF v4.2 , as this version supports the ESP32-S2?
If not, it would be great to get it running :)
Could somebody give info?

Regards

shahpiyushv, 2020-05-19:
> Hi,
> Is this repo compatible with ESP-IDF v4.2 , as this version supports the ESP32-S2?
> If not, it would be great to get it running :)
> Could somebody give info?
> 
> Regards

I just tried this on esp32-s2 with the esp-idf master branch. I had to make these changes to make it work.

```
diff --git a/examples/subscribe_publish/main/subscribe_publish_sample.c b/examples/subscribe_publish/main/subscribe_publish_sample.c
index c5b48ae..5772982 100644
--- a/examples/subscribe_publish/main/subscribe_publish_sample.c
+++ b/examples/subscribe_publish/main/subscribe_publish_sample.c
@@ -37,8 +37,10 @@
 #include "esp_wifi.h"
 #include "esp_event_loop.h"
 #include "esp_log.h"
+#ifdef CONFIG_EXAMPLE_SDCARD_CERTS
 #include "esp_vfs_fat.h"
 #include "driver/sdmmc_host.h"
+#endif
 
 #include "nvs.h"
 #include "nvs_flash.h"
@@ -328,5 +330,5 @@ void app_main()
     ESP_ERROR_CHECK( err );
 
     initialise_wifi();
-    xTaskCreatePinnedToCore(&aws_iot_task, "aws_iot_task", 9216, NULL, 5, NULL, 1);
+    xTaskCreatePinnedToCore(&aws_iot_task, "aws_iot_task", 9216, NULL, 5, NULL, 0);
 }
```

Schuck84, 2020-05-20:
Hi,
I managed to get a connection, but something runs into some issues as followed.
Could you have a look and maybe tell me how to fix it?

I (4037) subpub: Connecting to AWS...
I (7777) subpub: Subscribing...
I (8117) subpub: Stack remaining for task 'aws_iot_task' is 3644 bytes
E (9417) subpub: An error occurred in the main loop.

abort() was called at PC 0x4008585b on core 0
0x4008585b: aws_iot_task at c:\users\thoma\documents\git\esp-aws-iot\examples\subscribe_publish\build/../main/subscribe_publish_sample.c:300 (discriminator 2)

Backtrace:0x40029ddf:0x3ffdfc90 0x4002a59d:0x3ffdfcb0 0x400307ee:0x3ffdfcd0 0x4008585b:0x3ffdfd40 0x4002a5a5:0x3ffe0ae0
0x40029ddf: panic_abort at C:/Users/thoma/Documents/git/esp-idf/components/esp_system/panic.c:330

0x4002a59d: esp_system_abort at C:/Users/thoma/Documents/git/esp-idf/components/esp_system/system_api.c:100

0x400307ee: abort at C:/Users/thoma/Documents/git/esp-idf/components/newlib/abort.c:46

0x4008585b: aws_iot_task at c:\users\thoma\documents\git\esp-aws-iot\examples\subscribe_publish\build/../main/subscribe_publish_sample.c:300 (discriminator 2)

0x4002a5a5: vPortTaskWrapper at C:/Users/thoma/Documents/git/esp-idf/components/freertos/xtensa/port.c:143

S31416M, 2020-11-20:
Hi, any update on V4 version availability? The below link is a dead one.

 https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta

shahpiyushv, 2021-07-23:
Closing this issue since v4_beta would not be supported because Amazon has deprecated it. Support for their newer releases is currently available on the [release/beta](https://github.com/espressif/esp-aws-iot/tree/release/beta) branch.

closed 2021-07-23 by shahpiyushv