espressif/esp-aws-iot #14: Update Time Out

sagar448, 2019-09-05:
I have adapted the aws code for my needs, everything works perfectly if I only have uptil 7 JsonStruct_t in the aws iot shadow add reported function. Once I surpass 7 and add the 8th JsonStruct handler I start getting update time out error. I've tried increasing the update time from 4 seconds to almost 10 seconds but it does nothing. Also even though it shows update time out the shadow update is being accepted on aws and I can see the updated parameters. 

Please help! This isnt really an error because everything is still working okay, but I just don't like seeing the update time out error message.

@shahpiyushv
@mahavirj

shahpiyushv, 2019-09-05:
@sagar448 , since Thing-Shadow basically uses MQTT publish/subscribe itself, can you try increasing the values for CONFIG_AWS_IOT_MQTT_TX_BUF_LEN and CONFIG_AWS_IOT_MQTT_RX_BUF_LEN? Probably the payload may be going beyond the default sizes?

sagar448, 2019-09-05:
@shahpiyushv  Thank you for your help. I do not think the TX buff length and the RX buff length are the problem as I have tried to increase them before. And also because if I was exceeding the buffer length it would not update the aws iot (and would throw an error).

BUT i tried to increase it anyways as you suggested and the problem persists. Any other suggestions would help a lot! Thanks!

shahpiyushv, 2019-09-06:
@sagar448 , RX buffer length indeed could be the problem as AWS IoT sends a lot of metadata in response to a Shadow Update. To check exact size, you can subscribe to `$aws/things/<ThingName>/#` MQTT Topic on AWS Console and check all data being exchanged. The huge response can be found on topic `$aws/things/<ThingName>/shadow/update/documents`. I increased CONFIG_AWS_IOT_MQTT_RX_BUF_LEN to 3072 and tested with 8 JsonStruct_t and it worked fine for me.

sagar448, 2019-09-06:
Hello @shahpiyushv 

Thank you for your prompt responses. Now I have 14 jsonStructs and accordingly I decided to set the Buffer lengths first to 6000 and then to 8192. Still I get update time out. I subscribed to the above mqtt topics and my data is approximately 4kB.

Im still getting update time out after changing the buff length, do you know what else it could be?

sagar448, 2019-09-06:
Hello @shahpiyushv,

Thank you so much for your help. I manage to fix the problem with your suggestion. I was changing the RX and TX buff lengths in the config.h file for aws but not from the menuconfig. I changed it in menuconfig and it worked. 

For everyone else who faces this error, change the shadow RX buff length as well as the AWS Iot mqtt tx and rx buff lengths in menuconfig AND in the config.h file. Thank you :)

closed 2019-09-06 by sagar448