espressif/esp-aws-iot #15: AWS_IOT_MQTT_TX_BUF_LEN limited to 65536 bytes

erki1993, 2019-10-01:
Why is the KConfig variable AWS_IOT_MQTT_TX_BUF_LEN limited to max 65536 bytes. We have a project that requires uploads of ~80000 bytes.

config AWS_IOT_MQTT_TX_BUF_LEN
    int "MQTT TX Buffer Length"
    default 512
    range 32 65536
    help
        Maximum MQTT transmit buffer size. This is the maximum MQTT
        message length (including protocol overhead) which can be sent.

        Sending longer messages will fail.

shahpiyushv, 2019-10-17:
@erki1993 , changed it to 128K (131072 bytes) as that is that max allowed by AWS IoT as per https://docs.aws.amazon.com/iot/latest/developerguide/limits-iot.html.

shahpiyushv, 2019-11-06:
Closing this issue as it has been addressed.

closed 2019-11-06 by shahpiyushv

dizcza, 2022-12-18:
@erki1993 could you share the code for publishing such a large (binary) payload to AWS IoT with ESP32?
thanks