espressif/esp-aws-iot #17: ESP32 + Alexa echo project

Paullake85, 2019-12-06:
Hello,
I have to develop a smart product based on ESP32 able to receive vocal commands from the Alexa Echo. I found on google many example projects based on Arduino and open source libraries, but I cannot use those environments for a market product.
- Is the esp-aws-iot a good starting point to develop the communication between Alexa Echo and my product? 
- Do I need to use the Amazon FreeRTOS?
- Is there any related example project base only on code published in Espressif repository?  

Thank you,
Paul

shahpiyushv, 2019-12-19:
@Paullake85 , if you intend to enable control via Alexa Smart Skills, it requires much more handling, in addition to AWS IoT, primarily
- Some User Management mechanism
- Some logic to associate users with devices
- APIs to authenticate user, discover devices and get/set the parameters from Alexa.
- Alexa Smart Skill

- esp-aws-iot just handles the device to cloud (MQTT broker) communication. Other pieces will still be required
- Amazon FreeRTOS will also offer only the device to cloud communication, similar to esp-aws-iot and does not provide any additional facility.
- As mentioned above, since enabling Alexa control requires many components, especially on the cloud side, we do not have any direct example available for that. But we have some internal projects to help enable this. You can get in touch with our sales team or add me on LinkedIn for additional info.

twlawrenceko, 2021-11-03:
> @Paullake85 , if you intend to enable control via Alexa Smart Skills, it requires much more handling, in addition to AWS IoT, primarily
> 
> * Some User Management mechanism
> * Some logic to associate users with devices
> * APIs to authenticate user, discover devices and get/set the parameters from Alexa.
> * Alexa Smart Skill
> 
> So, answering your specific queries:
> 
> * esp-aws-iot just handles the device to cloud (MQTT broker) communication. Other pieces will still be required
> * Amazon FreeRTOS will also offer only the device to cloud communication, similar to esp-aws-iot and does not provide any additional facility.
> * As mentioned above, since enabling Alexa control requires many components, especially on the cloud side, we do not have any direct example available for that. But we have some internal projects to help enable this. You can get in touch with our sales team or add me on LinkedIn for additional info.

Hello @shahpiyushv ,
Sorry for commenting on an old thread, but I have questions regarding to using this example to map with smart home skills with Alexa.

Currently, Lambda function is set and Alexa Smart Home Skills APIs are done. The part that is unclear to me is how to enable discovery protocol on ESP itself. The discovery JSON is well defined on Lambda but the link between ESP and Alexa are missing. The other method is to make ESP update status to AWS IoT and update back to Alexa.

If an internal projects are available, can you provide the sales team for an example?

Thank you.

shahpiyushv, 2021-11-03:
@twlawrenceko , Alexa Smart Home Skills discovery sends a query to the vendor cloud and this cloud service creates and passes the appropriate JSON payload back to Alexa. The device itself is not directly involved anywhere in the communication.

Our [ESP RainMaker](https://rainmaker.espressif.com/) solution does something similar. The ESP nodes send data to cloud as per the format [here](https://rainmaker.espressif.com/docs/node-cloud-comm.html). The cloud service exposes all this info via [REST APIs](https://swaggerapis.rainmaker.espressif.com/#/User%20Node%20Association/getUserNodes), and the ESP RainMaker Alexa skill uses these APIs and reformats the data as per the Alexa specs.