espressif/esp-aws-iot #18: Subscribed payloads keep repeating when subscribed with QOS1 .

akshayvernekar, 2019-12-24:
Our usecase expects us to subscribe and publish with QOS of 1 , when the device subscribes with QOS 1 the payloads published to the subscribed topic keep repeating , it looks like the library is not sending the ACK message to the message published and the broker is trying to resend the messages until it gets Acknowledgment. This issue happens randomly the reproducibility rate is 1/5 times. 

I tried the latest [C sdk from aws ](https://github.com/aws/aws-iot-device-sdk-embedded-C)on my computer and I didnt find this issue , may I know if there is a release planned with the latest aws sdk ?

shahpiyushv, 2020-03-11:
@akshayvernekar , we will check what exactly is happening, but for the time being, can you check if the "isDup" parameter of IoT_Publish_Message_Params tells you if the message was a duplicate? For QOS1, it is indeed possible to get the same message multiple times.

mohinishnmrinq, 2020-03-13:
I have been facing the same problem of repeated subscribe payload with QOS1. I tried publishing payload every 5 sec with incrementing seq no within payload and I have noticed that out of 100 msgs I received 50 duplicate msgs. I also ran the same test using latest stable aws sdk and I received 2 duplicates out of 250 msgs. Getting duplicates with such margin doesn't seem to be ok.

Is there any way to fix this?

shahpiyushv, 2020-03-16:
@mohinishnmrinq , what are your observations with older aws sdk on raspberry pi? Depending on whether the behaviour is similar to esp-aws-iot or not, we can then check if we need to just update the aws sdk or check in our porting layer.

VivekPalle, 2021-11-23:
Hi, Is this issue closed or still in open. We are also facing the same issue. Using the AWS IOT example from SDK-v3.3.1.
Please suggest, if there is any solution available. Thanks.

shahpiyushv, 2021-11-23:
@VivekPalle , have you tried using QOS0 for publishing and also checking for the isDup flag mentioned in the [comment above](https://github.com/espressif/esp-aws-iot/issues/18#issuecomment-597581313)? Getting duplicates is indeed a possibility with QOS1 as network conditions can result in PUBACK not reaching the broker.

VivekPalle, 2021-11-26:
@shahpiyushv 
Yes, we tried QOS0 but some published events are missing. So, that's the reason we want to make it work with QOS1.
We also tried to check the isDUP flag but it is always zero for normal messages and the same for duplicate messages also. No change was found in the isDUP value.
I am just sharing the log file and screenshots for your reference. Please suggest what to do.
[isDUP value with QOS1-log.txt](https://github.com/espressif/esp-aws-iot/files/7606778/isDUP.value.with.QOS1-log.txt)
[image]

VivekPalle, 2021-11-29:
@shahpiyushv 
Do you have any other solution to try?

florjandule, 2022-09-22:
Is there any update on this topic? Setting the isDup as true when received first message doesn't work either.