espressif/esp-aws-iot #19: Test Issue

mahavirj, 2019-12-31:
Test Issue

closed 2019-12-31 by mahavirj

Horaira, 2020-01-01:
Hi Mahavir,
Just need some help regarding Integrating  Azure IoT SDK with our product which has FreeRTOS.
Can you please ping me on horairajilani.khan@in.bosch.com
It would be great help

mahavirj, 2020-01-02:
> Hi Mahavir,
Just need some help regarding Integrating Azure IoT SDK with our product which has FreeRTOS.
Can you please ping me on horairajilani.khan@in.bosch.com
It would be great help

Have initiated an email thread for this. Thanks.

Horaira, 2020-01-02:
> 
> 
> > Hi Mahavir,
> > Just need some help regarding Integrating Azure IoT SDK with our product which has FreeRTOS.
> > Can you please ping me on [horairajilani.khan@in.bosch.com](mailto:horairajilani.khan@in.bosch.com)
> > It would be great help
> 
> Have initiated an email thread for this. Thanks.

Thanks Mahavir !