espressif/esp-aws-iot #22: esp-aws-iot crashes work on esp8266

akshayvernekar, 2020-03-07:
Hi,
I am working on a project which involves connecting esp8266 to AWS IOT, I am using the HEAD version of ESP8266_RTOS_SDK  and using esp-aws-IOT as an external component  . After flashing the firmware I get "Task watchdog got triggered." error followed by a crash . I tried disabling task watchdog but this causes some wifi related issues . Following is the crash log, kindly guide 

```

> I (3097) mqtt_controller: Connecting to AWS...clientid: 1234
> I (3107) mqtt_controller: stage2
> Task watchdog got triggered.
> 
> Guru Meditation Error: Core  0 panic'ed (unknown). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x402638f0  PS      : 0x00000030  A0      : 0x402468ba  A1      : 0x3fff46f0  
> 0x402638f0: mpi_mul_hlp at /home/akki/esp/axsiot_base/components/mbedtls/mbedtls/library/bignum.c:1364 (discriminator 2)
> 
> 0x402468ba: mpi_montmul at /home/akki/esp/axsiot_base/components/mbedtls/mbedtls/library/bignum.c:737
> 
> A2      : 0x00000021  A3      : 0x3fffc208  A4      : 0x3fffe6a4  A5      : 0xa9cc47da  
> A6      : 0x560f939c  A7      : 0x0f2ec1b0  A8      : 0x0f2f6977  A9      : 0x00000000  
> A10     : 0xa67b8d12  A11     : 0xba280000  A12     : 0x11c80000  A13     : 0xbe000000  
> A14     : 0x0000000c  A15     : 0x4609f695  SAR     : 0x00000020  EXCCAUSE: 0x3fffe69c  
> 
> Backtrace: 0x402638f0:0x3fff46f0 
> 0x402638f0: mpi_mul_hlp at /home/akki/esp/axsiot_base/components/mbedtls/mbedtls/library/bignum.c:1364 (discriminator 2)
> 
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 7168, room 16 
> tail 0
> chksum 0x90
> load 0x3ffe8408, len 24, room 8 
> tail 0
> chksum 0x19
> load 0x3ffe8420, len 3532, room 8 
> tail 4
> chksum 0x03

```

Regards
Akshay

AnhTuDo1998, 2020-05-24:
Hi, looks like your program run into an infinite loop somewhere in your interrupt handler. You can check their documentation and maybe use the debugger?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html

shahpiyushv, 2020-05-24:
@akshayvernekar , we recently found an issue which was causing an incorrect default value of task watchdog timer, leading to such crashes. It has been fixed here: https://github.com/espressif/ESP8266_RTOS_SDK/commit/f03e81b0927fd0eabfc996692c01e440aa251d63

Please check if it helps. Sorry for the late reply.