espressif/esp-aws-iot #23: AWS IOT core OTA

AQIBRIAZ, 2020-04-20:
Hi there,

Thank you for the Library. Can you provide example for the OTA with AWS IOT core? 

Regards

shahpiyushv, 2020-04-21:
@AQIBRIAZ , if you are looking for OTA over MQTT, then we do not have any example for that. Normally, we prefer hosting the firmware image somewhere (say AWS S3) and then sending the URL to the device over MQTT. If that approach is fine, the relevant documentation and example can be found in ESP Jumpstart (https://docs.espressif.com/projects/esp-jumpstart/en/latest/firmwareupgrade.html)

AQIBRIAZ, 2020-04-23:
@shahpiyushv  Thanks for quick response. Can you guide me how to do the OTA securely with AWS S3 as you mentioned? I have done the OTA using AWS S3 but we need to make the OTA link public to do OTA.  Can you guide me the steps need to be done if we go with MQTT?

Thanks

shahpiyushv, 2020-04-23:
@AQIBRIAZ , there are two possible options

1. Use AWS IoT certificates to authorise the device to access other AWS services (https://docs.aws.amazon.com/iot/latest/developerguide/authorizing-direct-aws.html). This seems a bit complicated though, and I haven't tried this myself.
2. Keep your S3 buckets private, and send a pre-signed URL to the device over MQTT (https://docs.aws.amazon.com/AmazonS3/latest/dev/ShareObjectPreSignedURL.html). This is somewhat easier compared to the above. You can set the URL validity as per your use case.

j-izaguirre, 2021-03-31:
Hi @shahpiyushv i use your suggestion of use OTA using AWS S3 and keep bucket private and send pre-signed URL to the device over MQTT, but i receive a error MQTT_RX_BUFFER_TOO_SHORT_ERROR when the subscribe to the message with the pre-signed URL. the message send over MQTT is something like that:

{
  "url": "https://yyyyyyyyy.s3.us-east-2.amazonaws.com/xxxxxxxxxxxx.bin?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Credential=AKZARG9L3AVTDZ623X2G%2F20210331%2Fus-east-2%2Fs3%2Faws4_request&X-Amz-Date=20210331T153812Z&X-Amz-Expires=3600&X-Amz-Signature=364487548e39278fcc32a71105ecd486a69fa0078a3d72bae8fcdb1328c7c752&X-Amz-SignedHeaders=host&x-amz-user-agent=aws-sdk-js%2F3.8.0%20os%2Fwin32%2F10.0.19042%20lang%2Fjs%20md%2Fnodejs%2F12.15.0%20api%2Fs3%2F3.8.0&x-id=GetObject"
}

where can i change this buffer size? or how can i solve that  problem?

Thanks..

j-izaguirre, 2021-03-31:
Thanks, I solve the problem change the value of MQTT RX Buffer Length to 1024 in menuconfig -> Component Config -> Amazon Web Services IoT Platform.