espressif/esp-aws-iot #27: Error(-12) connecting to ac6aacx09z5rv-ats.iot.us-east- 2.amazonaws.com:8883[0m

nguyentranho, 2020-07-11:
I'm constantly running into this error without any clue. Could you give some hints?

shahpiyushv, 2020-07-13:
@nguyentranho , can you give some more details? Especially these:

- Have you created the thing and certificate and attached appropriate policy?
- Have you verified that everything works fine with other MQTT clients (say mosquitto) or direct [AWS IoT HTTP APIs](https://docs.aws.amazon.com/iot/latest/developerguide/http.html) from a host machine in the same network?
- Do you always see this error or is it intermittent?

nguyentranho, 2020-07-15:
Hi, I have fixed it with following policy. I'm still confused about its syntax. I was trying to explicitly set the filter to "topic/esp32/sub" but it always replies with -12 error
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "arn:aws:iot:us-east-2:438468228341:*"
    }
  ]
}

shahpiyushv, 2020-07-18:
@nguyentranho , you can use wildcards in your resources, but if you want the policy to be more restrictive, for Publish and Receive, the Resource is in this format: `"arn:aws:iot:*:*:topic/<your-topic>"`. For subscribe, it is `"arn:aws:iot:*:*:topicfilter/<your-topic>"`, and for connect, it is `"arn:aws:iot:*:*:client/<client-id>"`

ujur007, 2021-07-20:
I had the similar issue, I solved by creating a policy and properly editing the resources in connect, publish and subscribe.