espressif/esp-aws-iot #31: how to stop triggering iot_subscribe_callback_handler when we call aws_iot_mqtt_publish function.

hariprasad58, 2020-08-25:
When we I call aws_iot_mqtt_publish function it is automatically triggering  iot_subscribe_callback_handler function, but I want to run  iot_subscribe_callback_handler function only sub task not for pub task, Please help me how to achieve it. Thanks in advance.

shahpiyushv, 2020-08-25:
@hariprasad58 , if your subscribe and publish topics are same, the callback will indeed get triggered as the broker will send back the message to the device. Can you confirm if they are same or different?

hariprasad58, 2020-08-25:
Hi @shahpiyushv yes you are correct, I am using same Topic name, should I change that to different to fix this issue..??

shahpiyushv, 2020-08-25:
@hariprasad58 , yes that's the simplest possible solution.

hariprasad58, 2020-09-12:
Hi @shahpiyushv Sorry for the late reply, thanks for the solution it is working fine, Just curious to know are there any other ways to fix this apart from this solution, Please let me know. Thanks in advance.