espressif/esp-aws-iot #35: Subscibed topic stops triggering  iot_subscribe_callback_handler after new subscripition

Mluis31, 2020-09-10:
For project reasons i have to subscribe some topics inside the while loop int e aws_iot_task. After i subscribe to a second set of topics the first set does not trigger iot_subscribe_callback_handler anymore. I also start receiving the messages of the second set of topics in duplicates. What may be the reason?

Mluis31, 2020-09-10:
Shure, here it is the aws_iot_task:

void aws_iot_task(void *param)
{
    char cPayload[100];

    AWS_IoT_Client client;

    char aux_client_id[20];
    sprintf(aux_client_id, "gateway-%s", get_address());
    CLIENT_ID = aux_client_id;

    IoT_Error_t rc = FAILURE;

    IoT_Client_Init_Params mqttInitParams = iotClientInitParamsDefault;
    IoT_Client_Connect_Params connectParams = iotClientConnectParamsDefault;

    //IoT_Publish_Message_Params paramsQOS0;
    //IoT_Publish_Message_Params paramsQOS1;

    ESP_LOGI(TAGm, "AWS IoT SDK Version %d.%d.%d-%s", VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH, VERSION_TAG);

    mqttInitParams.enableAutoReconnect = false; // We enable this later below
    mqttInitParams.pHostURL = HostAddress;
    mqttInitParams.port = port;

    mqttInitParams.pRootCALocation = (const char *)aws_root_ca_pem_start;
    mqttInitParams.pDeviceCertLocation = (const char *)certificate_pem_crt_start;
    mqttInitParams.pDevicePrivateKeyLocation = (const char *)private_pem_key_start;

    mqttInitParams.mqttCommandTimeout_ms = 20000;
    mqttInitParams.tlsHandshakeTimeout_ms = 5000;
    mqttInitParams.isSSLHostnameVerify = true;
    mqttInitParams.disconnectHandler = disconnectCallbackHandler;
    mqttInitParams.disconnectHandlerData = NULL;

    ESP_LOGI(TAGm, "init");

    rc = aws_iot_mqtt_init(&client, &mqttInitParams);
    if (SUCCESS != rc)
    {
        ESP_LOGE(TAGm, "aws_iot_mqtt_init returned error : %d ", rc);
        abort();
    }

    char aux_dis[27];
    sprintf(aux_dis, "gw/%s/discon", get_address());
    char *TOPIC_DIS = aux_dis;
    const int TOPIC_DIS_LEN = strlen(TOPIC_DIS);

    IoT_MQTT_Will_Options lwtParams;

    sprintf(cPayload, "%s", "");

    lwtParams.pTopicName = TOPIC_DIS;
    lwtParams.topicNameLen = TOPIC_DIS_LEN;
    lwtParams.qos = 1;
    lwtParams.pMessage = cPayload;
    lwtParams.msgLen = strlen(cPayload);
    lwtParams.isRetained = false;

    if (get_device_type() == 'w') /* Wait for WiFI to show as connected */
        xEventGroupWaitBits(wifi_event_group, getConnectBit(), false, true, portMAX_DELAY);
    else if (get_device_type() == 'g') /* Wait for WiFI to show as connected */
        xEventGroupWaitBits(gsm_event_group, getConnectBit(), false, true, portMAX_DELAY);
    
    green_led_on();
    vTaskDelay(100 / portTICK_RATE_MS);

    connectParams.keepAliveIntervalInSec = 10;
[... 190 lines cut ...]

        if (get_check_ping())
        {
            check_devices_pings(&client);
            set_check_ping(false);
        }

        if (get_prc_msg())
        {
            process_message(&client, get_type(), get_sender(), get_payload(), get_sig());
            set_type(0);
            set_sender("");
            set_payload("");
            set_sig(0);
            set_prc_msg(false);
        }

        //ESP_LOGI(TAGm, "Stack remaining for task '%s' is %d bytes", pcTaskGetTaskName(NULL), uxTaskGetStackHighWaterMark(NULL));

        vTaskDelay(1000 / portTICK_RATE_MS);
    }

    ESP_LOGE(TAGm, "An error occurred in the main loop. RC: %d", rc);
    abort();
}

shahpiyushv, 2020-09-10:
@Mluis31 Can you also specify which are the topics that stop getting callbacks and which ones are getting twice?

Mluis31, 2020-09-10:
First it subscribes to the topic sd/id/+ and sd/id/++ in here:
if (get_sub_d())
    {
        //mqtt_subscribe_to_device_topics(&client, get_sub_device());
        //SUBSCRIBE TO TOPICS:
        ESP_LOGI(TAGm, "Subscribing to device \"%s\" topics: ", get_sub_device());

        //IoT_Error_t rc = FAILURE;

        char aux[20];
        sprintf(aux, "sd/%s", get_sub_device());
        const char *TOPIC_SUB = aux;

        char aux_sub[30];
        sprintf(aux_sub, "%s/+", TOPIC_SUB);
        const char *TOPIC_all = aux_sub;
        const int TOPIC_all_LEN = strlen(TOPIC_all);
        printf("%s\n", TOPIC_all);

        char aux_sub1[30];
        sprintf(aux_sub1, "sd/%s/+/+", get_sub_device());
        const char *TOPIC_all1 = aux_sub1;
        const int TOPIC_all1_LEN = strlen(TOPIC_all1);

        printf("%s\n", TOPIC_all1);

        /*---------------------------------------------* //DONE
        *    SUBSCRIBE TOPICS TO LISTEN
        *---------------------------------------------*/
        //ESP_LOGI(TAGm, "Subscribing...");
        rc = aws_iot_mqtt_subscribe(&client, TOPIC_all, TOPIC_all_LEN, QOS0, iot_subscribe_callback_handler, NULL);
        if (SUCCESS != rc)
        {
            ESP_LOGE(TAGm, "Error subscribing : %d ", rc);
            abort();
        }
        else
            ESP_LOGI(TAGm, "SUCCESS");

        //ESP_LOGI(TAGm, "Subscribing...");
        rc = aws_iot_mqtt_subscribe(&client, TOPIC_all1, TOPIC_all1_LEN, QOS0, iot_subscribe_callback_handler, NULL);
        if (SUCCESS != rc)
        {
            ESP_LOGE(TAGm, "Error subscribing : %d ", rc);
            abort();
        }
        else
            ESP_LOGI(TAGm, "SUCCESS");
        set_sub_d(false);
        //vTaskDelay(100 / portTICK_RATE_MS);
        mqtt_request_device_info(&client, get_sub_device());
    }

Than it subscribes to the topic g/id/+ in here: 
if (get_sub_g())
    {
        //mqtt_request_group_info(&client, get_sub_group());
        ESP_LOGI(TAGm, "Resquest group info: ");

        //SUBSCRIBE FOR RESPONSE:
        char aux_g[10];
        sprintf(aux_g, "g/%s", get_sub_group());
        const char *TOPIC_G_Prefix = aux_g;

        char aux_gi[20];
        sprintf(aux_gi, "%s/+", TOPIC_G_Prefix);
        const char *TOPIC_G_Info = aux_gi;
        const int TOPIC_G_Info_LEN = strlen(TOPIC_G_Info);
        
        ESP_LOGI(TAGm, "Subscribing...%s", TOPIC_G_Info);
        rc = aws_iot_mqtt_subscribe(&client, TOPIC_G_Info, TOPIC_G_Info_LEN, QOS0, iot_subscribe_callback_handler, NULL);
        if (SUCCESS != rc)
        {
            ESP_LOGE(TAGm, "Error subscribing : %d ", rc);
            abort();
        }
        else
            ESP_LOGI(TAGm, "SUCCESS");

        set_sub_g(false);
    }

After this it stops gettings callbacks from sd/id/+ and sd/id/+/+ and starts getting callbacks from g/id/+ in duplicate

shahpiyushv, 2020-09-10:
@Mluis31 , I am not 100% sure if this could be the issue, but the subscription topics' buffer generally need to stay allocated throughout the lifetime of the subscription. In your case, you are using variables defined inside if() blocks for the topics, which would not be valid outside the block. They may get overwritten by other variables outside the block. Can you try defining all those variables outside the blocks or explicitly allocating memory for them?

Mluis31, 2020-09-14:
Ok i will try that and get back to you.

Mluis31, 2020-09-14:
@shahpiyushv , Now i can get the the callback for every topic. But i still get the topic in duplicate.

shahpiyushv, 2020-09-14:
@Mluis31 duplicates for all the topic or just the latter as you mentioned earlier?

Mluis31, 2020-09-14:
all topics @shahpiyushv .

Mluis31, 2020-09-16:
Do you know what it might be? @shahpiyushv

shahpiyushv, 2020-09-17:
@Mluis31 it is hard to tell. Will have to dig deeper into how the underlying aws-iot-device-sdk-embedded-C handles callbacks. Since your subscriptions are QOS0, we can be sure that the broker isn't sending the message twice.

Mluis31, 2020-09-29:
Does anyone have any tips for this problem?

shahpiyushv, 2020-09-29:
@Mluis31 , I will check again when I get time, but can you also ask this on  [aws-iot-device-sdk-embedded-C](https://github.com/aws/aws-iot-device-sdk-embedded-C) since the publish/subscribe logic is in the AWS code?

Mluis31, 2020-09-29:
ok thank you