espressif/esp-aws-iot #37: Using aws-iot shadow along with publish on topic causing tls error -0x4e (IDFGH-3977)

rhr407, 2020-09-17:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version:  v4.1
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am using aws-iot component from [here](https://duckduckgo.comhttps://github.com/espressif/esp-aws-iot) and added to the components folder of esp-idf. I want to use both aws-iot shadow service along with publishing some data to another topic. The problem is that I am facing an mbedtls error -0x4e. Sometimes, aws-shadow disconnects as well. I would be grateful if you kindly help me with this issues.

### Expected Behavior

These should be no errors in publishing the message to aws topic along with running shadow in parallel. Both threads should work smoothly.

### Actual Behavior
Th

### Steps to reproduce

Just use the two components of aws-iot i.e. aws-iot shadow and pubsub in parallel as separate threads in app_main

### Code to reproduce this issue

## Debug Logs

```
E (340173) aws_iot: failed! mbedtls_ssl_write returned -0x4e
Published: {"DT":2,"T":53,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}I (340273) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5

I (340273) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned -26624 (-0x6800)

W (340283) mbedtls: ssl_tls.c:4973 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (340293) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -26624 (-0x6800)

W (340303) mbedtls: ssl_tls.c:8335 mbedtls_ssl_read_record() returned -26624 (-0x6800)

I (340323) mbedtls: ssl_tls.c:8270 => read

I (340323) mbedtls: ssl_tls.c:4311 => read record

I (340323) mbedtls: ssl_tls.c:2536 => fetch input

I (340333) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (340433) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5

I (340433) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned -26624 (-0x6800)

W (340433) mbedtls: ssl_tls.c:4973 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)

W (340443) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -26624 (-0x6800)

W (340453) mbedtls: ssl_tls.c:8335 mbedtls_ssl_read_record() returned -26624 (-0x6800)

I (340473) mbedtls: ssl_tls.c:8270 => read

I (340473) mbedtls: ssl_tls.c:4311 => read record

I (340473) mbedtls: ssl_tls.c:2536 => fetch input

I (340483) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (340583) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5
[... 319 lines cut ...]
I (343173) mbedtls: ssl_tls.c:4311 => read record

I (343173) mbedtls: ssl_tls.c:2536 => fetch input

I (343183) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5

I (343193) mbedtls: ssl_tls.c:8682 => write

I (343193) mbedtls: ssl_tls.c:2755 => flush output

I (343193) mbedtls: ssl_tls.c:2774 message length: 168, out_left: 168

I (343203) mbedtls: ssl_tls.c:2779 ssl->f_send() returned -78 (-0x004e)

W (343213) mbedtls: ssl_tls.c:8612 mbedtls_ssl_flush_output() returned -78 (-0x004e)

I (343223) mbedtls: ssl_tls.c:8710 <= write

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ncohen17, 2020-09-28:
Hello,
Any updates on fixing this issue ? Is there a solution to this issue ?

ajyta, 2020-09-30:
Same problem here on ESP32CAM Arduino IDE :-(

aleblsv, 2022-01-20:
@shahpiyushv Is there an example for combining both shadow with publish subscribe?

shahpiyushv, 2022-01-21:
@aleblsv , no there isn't. But you can use the `aws_iot_mqtt_subscribe()` and `aws_iot_mqtt_publish()` calls with the same mqttClient as you see in [aws_iot_shadow_connect()](https://github.com/espressif/esp-aws-iot/blob/249c573cbdcf653c03a6bf3a30f658f005856880/examples/thing_shadow/main/thing_shadow_sample.c#L266).

aleblsv, 2022-01-22:
@shahpiyushv 
Thanks it working for me,
1. I took  connection parameters from the shadow example 
2. Use same mqttClient for aws_iot_mqtt_subscribe() and aws_iot_mqtt_publish() and shadow
3. Only aws_iot_shadow_yield() in the while loop