espressif/esp-aws-iot #38: Aws disconnection crash or loops

romocitto88, 2020-09-30:
Hi, I've a similar issue: if I try to connect and disconnect for n times from AWS Mqtt broker, sometimes the esp32 module reset when is executed "aws_iot_mqtt_disconnect(&hdl_mqtt_Client);" or loops into the function and never returns.

The error:

> Disconnect Mqtt
>C:\.platformio\packages\framework-espidf\components\freertos\queue.c:720 (xQueueGenericSend)- assert failed!
>abort() was called at PC 0x40098a7c on core 1
>
>ELF file SHA256: f997dde20128e6df
>
>Backtrace: 0x4008da79:0x3ffe2f60 0x4008ddfd:0x3ffe2f80 0x40098a7c:0x3ffe2fa0 0x4012f822:0x3ffe2fe0 0x40133210:0x3ffe3000 >0x4013331d:0x3ffe3030 0x40135dde:0x3ffe3050 0x40136801:0x3ffe3080 0x40130445:0x3ffe30a0 0x401304ec:0x3ffe30c0 >0x40098381:0x3ffe30f0
>
>Rebooting...
>

Decoding stack results:

>0x40133185: select_check_waiters at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\api\sockets.c line 2672
>0x4013331d: event_callback at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\api\sockets.c line 2623
>0x40135fa1: recv_tcp at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\api\api_msg.c line 351
>0x40124d2d: tcp_input at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\core\tcp_in.c line 501
>0x40129e16: ip4_input at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\core\ipv4\ip4.c line 784
>0x4012f6b2: ethernet_input at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\netif\ethernet.c line 186
>0x4013046d: tcpip_thread_handle_msg at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\api\tcpip.c line 180
>0x401304ec: tcpip_thread at C:\.platformio\packages\framework-espidf\components\lwip\lwip\src\api\tcpip.c line 154
>0x40098381: vPortTaskWrapper at C:\.platformio\packages\framework-espidf\components\freertos\port.c line 143

The code:

```
void hdl_mqtt_deinit(void)
{
    printf("Delete Task\n");
    vTaskDelete(hdl_mqtt_taskHandler); //Delete the mqtt main task with yeld function
    printf("Disconnect Mqtt\n");
    IoT_Error_t err = aws_iot_mqtt_disconnect(&hdl_mqtt_Client); //Disconnect from AWS
    if(err!=SUCCESS)
        printf("Disc Unsuccess\n");
    else
        printf("Disc Success\n");
    err = aws_iot_mqtt_free(&hdl_mqtt_Client); //Free memory
    if(err!=SUCCESS)
        printf("free Unsuccess\n");
    else
        printf("free Success\n");
}
```

_Originally posted by @romocitto88 in https://github.com/espressif/esp-aws-iot/issues/29#issuecomment-699891789_

shahpiyushv, 2020-09-30:
@romocitto88 , we will check this, but calling `vTaskDelete()` directly isn't really recommended, especially when the task being deleted could be in the middle of something. A better approach would be to signal the task and then let it finish cleanly and then call `vTaskDelete(NULL);`

romocitto88, 2020-10-02:
Hi @shahpiyushv thank you for the suggestion. I will try to change it and do some tests. 
After that I will write a feedback.