espressif/esp-aws-iot #40: aws iot sdk v4.0 integration in esp-idf v3.3 and esp-idf v4.1 (IDFGH-4072)

chaabanemalki, 2020-10-21:
Hello,

AWS released few days ago the aws iot 4.0 and there were a branch with aws 4.0 beta in the esp-idf now it is removed.

Can you please share your timeline about the integration of aws IoT 4.0 into esp-idf 4.1 and v3.3

Thank you,
Chaabane

AshUK, 2021-01-08:
@shahpiyushv any chance we can update/estimate on the port for 202012.x ?

It looks to be using the FreeRTOS coreMQTT client and some other AWS FreeRTOS components internally

@lhespress already has a port of FreeRTOS Core MQTT as part of https://github.com/espressif/esp-idf/issues/6112#

petergzmo, 2021-02-02:
@shahpiyushv is there any update on this effort?  We would like to make use of the nicely integrated OTA functions of AWS while avoiding Amazon FreeRTOS, which is not feature complete and is difficult to develop with.