espressif/esp-aws-iot #41: When to support aws custom authentication？

HaydnLiao, 2020-12-03:
[Connecting to AWS IoT Core by using custom authentication](https://docs.aws.amazon.com/iot/latest/developerguide/custom-auth.html)
[Support MQTT username/password authentication](https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/641)
I want to test this feature using ESP32.

HaydnLiao, 2020-12-14:
@brucejcooper @projectgus @kedars @mahavirj

shahpiyushv, 2020-12-18:
@HaydnLiao , the [pointer](https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/641) you have shared applies to the revamped aws-iot-device-sdk-embedded-C. We haven't yet started porting work for that, and so cannot give an exact timeline right now. However, since the revamped SDK has now been stable for quite some time, we will prioritise the porting activity. I will provide an update here once we have better clarity about the efforts required.