espressif/esp-aws-iot #47: aws-iot-device-sdk-embedded-C 2012xx.xx support in esp idf v4.2

rmshub, 2021-03-11:
Hi Team,

I'm looking for IOT jobs for OTA updates. This feature are introduced in newer aws-iot-device-sdk-embedded-C.
Will be there support for latest aws-iot-device-sdk-embedded-C as esp-idf component

rmshub, 2021-03-13:
Hi Team,

Is info on this request?

samuelbles07, 2021-03-27:
I guess this project is discontinued?

shahpiyushv, 2021-04-08:
We are working on adding the required pieces one by one, starting with plain publish/subscribe followed by thing-shadow and later OTA, Jobs, etc. You may expect updates to this repo starting next week.

rmshub, 2021-04-09:
Thanks for update

Prabhuelectro, 2021-08-04:
@shahpiyushv no updates, now it is august.

shahpiyushv, 2021-08-04:
@Prabhuelectro @rmshub , sorry we did not update here, but as you can see in my comment [here](https://github.com/espressif/esp-aws-iot/issues/53#issuecomment-880927892), beta releases for the same are already available.  You may expect further updates on this in a day or two.

rmshub, 2021-11-17:
When will be esp-aws-iot beta version turned to stable release.

rmshub, 2021-11-27:
Any update on this topic ?

jeffxor, 2022-03-14:
Any update on this? Is this project discontinued?

No commits for 9 months and using a library that is over 2 years old. When can we expect some updates?

closed 2024-01-31 by rmshub