espressif/esp-aws-iot #48: device certificate verification fail

asurti9, 2021-04-08:
Hello all,

I am facing issue while connecting my esp32s2 device with the AWS cloud. I have followed below steps:
1. I have created my own root CA and registered in the AWS using below steps:
https://docs.aws.amazon.com/iot/latest/developerguide/register-CA-cert.html
2. Created device certificate using the above generated root CA and registered it with my IoT thing. Followed below link for the same.
https://docs.aws.amazon.com/iot/latest/developerguide/manual-cert-registration.html
3. Now I am using that device certificate, device private key and generated root CA for communicating with AWS by placing it on the below path:
https://github.com/espressif/esp-aws-iot/tree/master/examples/thing_shadow/main/certs
Built it and flashed the bins, the device is trying to communicate with the AWS cloud. But the TLS is failing. Attached the logs for the reference.
[fail.log](https://github.com/espressif/esp-aws-iot/files/6278280/fail.log)

Is there something or some steps which I am missing?

Thanks in advance!

shahpiyushv, 2021-04-08:
@asurti9 , The rootCA to be placed in the certs/ folder should not be the one created by you, but the AWS ROOT CA, which you can find [here](https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html#server-authentication-certs). Use [AmazonRootCA1.pem](https://www.amazontrust.com/repository/AmazonRootCA1.pem).

Moreover, you would also have to [attach a policy](https://docs.aws.amazon.com/iot/latest/developerguide/attach-to-cert.html) to your certificate. You can find some examples [here](https://docs.aws.amazon.com/iot/latest/developerguide/example-iot-policies.html).

asurti9, 2021-04-08:
Hey @shahpiyushv 

Thanks for the reply. With AWS ROOT CA and AWS device certificate (one-click certificate) the thing shadow example is working fine.
But we need to use our own generated device certificate signed by our generated ROOT CA which is already registered in AWS. This is the use case which we want to implement. But during this, it fails to verify the certificate.
So can you please help me with how can we achieve that? Why device certificate verify is failing? 

Thanks again!

shahpiyushv, 2021-04-08:
@asurti9 , the process you have followed earlier is fine. However, the usage of Root CA was wrong. The Root CA you created for signing the certificates is actually used by AWS IoT to authenticate the clients. The server certificate to be copied in the certs/ folder is something used by clients to authenticate the server (This is the X.509 mutual authentication).

Irrespective of what you use to sign your client certificates, this root CA certificate would always be the [AmazonRootCA1.pem ](https://www.amazontrust.com/repository/AmazonRootCA1.pem) mentioned in my comment.

asurti9, 2021-04-09:
Hello @shahpiyushv ;
We tried placing the cert as suggested by you and seems that the tls handshaking is now completed. However, the device is getting stuck when it is trying to connect to the endpoint.
As per the logs, it is visible that the tls handshaking is completed as the client state is 15. After that, it is trying to connect to the shadow and giving the error "-12" as seen in the below logs:

I (16619) mbedtls: ssl_tls.c:9122 <= free
V (16619) aws_iot: FUNC_ENTRY:   aws_iot_mqtt_set_client_state L#139 
V (16629) aws_iot: FUNC_ENTRY:   aws_iot_mqtt_client_lock_mutex L#85 
V (16629) aws_iot: FUNC_EXIT:   aws_iot_mqtt_client_lock_mutex L#103 Return Code : 0 
V (16639) aws_iot: FUNC_ENTRY:   aws_iot_mqtt_get_client_state L#67 
V (16649) aws_iot: FUNC_EXIT:   aws_iot_mqtt_get_client_state L#72 Return Code : 2 
V (16659) aws_iot: FUNC_EXIT:   aws_iot_mqtt_set_client_state L#164 Return Code : 0 
V (16659) aws_iot: FUNC_EXIT:   aws_iot_mqtt_connect L#484 Return Code : -12 
V (16669) aws_iot: FUNC_EXIT:   aws_iot_shadow_connect L#133 Return Code : -12 
E (16679) shadow: aws_iot_shadow_connect returned error -12, aborting...
abort() was called at PC 0x400882f3 on core 0

Attaching the detailed logs for more reference. The -12 error indicates that there is SSL read error. So just wanted to know if we are on the right track and if possible can you guide us where to look for this error?

Thanks!
[09.04.cloud.conn.fail.with.port.443.log](https://github.com/espressif/esp-aws-iot/files/6285607/09.04.cloud.conn.fail.with.port.443.log)

shahpiyushv, 2021-04-09:
@asurti9 , have you attached the appropriate AWS IoT policy? Can you share it here if possible? A simple, but extremely insecure policy is as below

```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "*",
      "Resource": "*"
    }
  ]
}
```

Note that you may use this just to verify that this is not an AWS IoT Policy issue. However, the actual policy should be much more stricter as per the docs [here](https://docs.aws.amazon.com/iot/latest/developerguide/example-iot-policies.html)

asurti9, 2021-04-12:
Thanks @shahpiyushv !
That helped, closing the issue now!

closed 2021-04-12 by asurti9