espressif/esp-aws-iot #52: Request to update to v3.15

MrutyunjayPatel, 2021-05-03:
hi , 
we are in the process of updating our projects for IDF 3.2.X to IDF 4.2.0 

Could we have the esp-aws-iot repo  update to latest compatible version from aws release repo.

I assume that esp-aws-iot is  currently is based on. 
https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/61f25f34712b1513bf1cb94771620e9b2b001970.
which is derived from V3.0.2 from the original aws sdk
https://github.com/aws/aws-iot-device-sdk-embedded-C

Last compatible version is  https://github.com/aws/aws-iot-device-sdk-embedded-C/releases/tag/v3.1.5 

Before esp-aws-iot moving to newer versions  based on 2020XX/2021XX as and when it will be available in the near future .

Thanks and regards. 
Mrutyunjay

austingrismer, 2021-05-17:
+1, @MrutyunjayPatel have you been able to get the library working at all in IDFv4.2.0? For me initialization/connection/enable-autoreconnect work fine but subscribing to a topic returns error code -13  (network disconnect) even though I am able to ping the device.

shahpiyushv, 2021-06-22:
@MrutyunjayPatel esp-aws-iot has now been updated to v3.1.5 of aws iot

MrutyunjayPatel, 2021-06-22:
Thanks @shahpiyushv

closed 2021-06-22 by MrutyunjayPatel