espressif/esp-aws-iot #53: AWS IoT sdk latest version(202103) as an esp-idf component

Devyami-Apcon, 2021-05-24:
Hey team, 

I am trying to integrate the latest version of AWS IoT SDK embedded C(v - 202103) to my project to leverage OTA feature. I don't prefer using Amazon Freertos as it limits my choice of ESP-IDF. It would be great if I could somehow include aws latest libraries like an esp-idf component.  

However in their latest libraries their demos are only compatible with posix integration. I am bit overwhelmed to integrate it. Need some help here. 

Are you people working on updating the current repo? If yes, how long would it take? 

Thanks

shahpiyushv, 2021-05-24:
@Devyami-Apcon The new SDK porting is work in progress and it is taking time because as you rightly said, it is somewhat overwhelming :). We may release a beta version by end of this week or next week, but the OTA feature would be available by mid June. The first release will just have publish/subscribe, thing-shadow and jobs.

Devyami-Apcon, 2021-05-25:
Really appreciate your prompt reply @shahpiyushv . 

Thanks for the update, we are eagerly waiting for this release :)

Devyami-Apcon, 2021-06-07:
Any update on the beta release @shahpiyushv ?

vamsi1105, 2021-07-05:
Hi @shahpiyushv is there any update on adding aws-iot as an esp-idf component ? We are waiting for the update.

pmumby, 2021-07-15:
Hi, just to chime in, we are developing the next generation of our existing commercial product on ESP32 (coming from our previous generation hardware device which was PIC32 based) which relies on OTA functionality, our deadline for global product availability (for roll-out in large scale) is October of this year... 

We've finished porting all of our firmware code over to ESP IDF 4.2 and everything is working beautifully, there is still some optimization and polish, as well as a few smaller less critical features we could work on, but the one remaining critical feature we need to implement is OTA capability (fully integrated into AWS IoT Core). 

@shahpiyushv if you could provide any update on when this version which supports the OTA capabilities would be greatly appreciated, so that I can decide if I focus out attention on writing OTA functionality from scratch, or if we can focus in other areas of our firmware code to buy time for this feature to become available.

Thanks!

shahpiyushv, 2021-07-15:
To answer the original issue, the basic mqtt and Thing Shadow support is now available in beta here:

https://github.com/espressif/esp-aws-iot/tree/release/beta

The OTA support is still work in progress, but the current status can be checked here: 

https://github.com/espressif/esp-aws-iot/tree/ota-beta/202103.00. This is not clean and finalised and so isn't yet ready for use. It can be considered more like preview