espressif/esp-aws-iot #56: How can I send data to Named shadow , not classic shadow

ESP3235, 2021-06-18:
Hi .
I have two shadows , classic and shadow1.
how can I connect to shadow1?
this example is connected to shadow classic only.

Thanks.

shahpiyushv, 2021-06-22:
@ESP3235 Support for Named shadows will be available soon, as we would make the new AWS IoT support available in a few days.

closed 2021-06-24 by ESP3235

twlawrenceko, 2021-12-15:
> @ESP3235 Support for Named shadows will be available soon, as we would make the new AWS IoT support available in a few days.

@shahpiyushv  Is the named shadows supported by now? I tried with `thing_shadow` example but still appears as classic shadow.

Thank you.

dhavalgujar, 2021-12-15:
Hi @twlawrenceko,
Named shadow [support is available](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/202103.00/CHANGELOG.md?plain=1#L9) in the AWS IoT Embedded C SDK 202103.00 release and up.

You can change [this macro](https://github.com/espressif/esp-aws-iot/blob/release/beta/examples/thing_shadow/main/demo_config.h#L263) in the thing_shadow example to use named shadows, and refer to the topics for named shadows in the AWS IoT documentation available [here](https://docs.aws.amazon.com/iot/latest/developerguide/reserved-topics.html#reserved-topics-shadow), for further use.

Hope this helps.