espressif/esp-aws-iot #63: CMake Error when compiling Example

FalconBFR, 2021-08-15:
Hello all! I followed the instructions of the readme but I am experiencing an error with CMake. I am a beginner but I believe it is something to do with the CMake configuration. Any suggestions are appreciated.

Full Error Log when building: (Does not contain sensitive info like keys)
```
> Executing task: cmake -G Ninja .. <

-- Found Git: /usr/local/bin/git (found version "2.23.0") 
-- Unexpected file in components directory: /Users/0220182218/esp/esp-idf/components/.DS_Store
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/0220182218/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/0220182218/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/0220182218/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/0220182218/esp/esp-idf/requirements.txt are satisfied.
CMake Error at /Users/0220182218/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at /Users/0220182218/esp/esp-idf/CMakeLists.txt:2 (project):

    project command is not scriptable

  Call Stack (most recent call first):

    /Users/0220182218/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:97 (include)
    /Users/0220182218/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:115 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /Users/0220182218/esp/esp-idf/tools/cmake/build.cmake:414 (__component_get_requirements)
  /Users/0220182218/esp/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/0220182218/esp/aws_esp32/subscribe_publish/build/CMakeFiles/CMakeOutput.log".
The terminal process "bash '-c', 'cmake -G Ninja ..'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: cmake --build . <

ninja: error: loading 'build.ninja': No such file or directory
The terminal process "bash '-c', 'cmake --build .'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: cmake --build . <

ninja: error: loading 'build.ninja': No such file or directory
The terminal process "bash '-c', 'cmake --build .'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: cmake --build . <

ninja: error: loading 'build.ninja': No such file or directory
The terminal process "bash '-c', 'cmake --build .'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: cmake --build . <

ninja: error: loading 'build.ninja': No such file or directory
The terminal process "bash '-c', 'cmake --build .'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

FalconBFR, 2021-08-15:
Update: I tried this on both Mac and Linux and the result is the same:
Here is the full output on Linux when building inside VSCode 
```
> Executing task: cmake -G Ninja .. <

-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/clement/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/clement/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/clement/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/clement/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/clement/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /home/clement/esp/esp-idf/requirements.txt are satisfied.
CMake Error at /home/clement/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  fatal: not a git repository (or any of the parent directories): .git

  fatal: not a git repository (or any of the parent directories): .git

  CMake Error at /home/clement/esp/esp-idf/CMakeLists.txt:2 (project):

    project command is not scriptable

  Call Stack (most recent call first):

    /home/clement/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/clement/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/clement/esp/esp-idf/tools/cmake/build.cmake:488 (__component_get_requirements)
  /home/clement/esp/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/home/clement/esp/aws-test/subscribe_publish/build/CMakeFiles/CMakeOutput.log".
The terminal process "bash '-c', 'cmake -G Ninja ..'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

dhavalgujar, 2021-08-15:
Hi @FalconBFR
I feel this issue might be specific to VS Code.
We will try building it inside VS Code, and reply here with what we find.

Meanwhile, can you try compiling it directly from the command line?
i.e.
By running `idf.py build` in the respective example's directory?

For example:
For `thing_shadow` example, try running the build command in `esp-aws-iot/examples/thing_shadow/`
For MQTT `tls_mutual_auth` example, try running the build command in `esp-aws-iot/examples/mqtt/tls_mutual_auth/`

FalconBFR, 2021-08-15:
@dhavalgujar Thanks for the quick reply! I tried building it with idf.py in linux and the issue still persists. Would a call when you are free or some screen recording/screenshots be helpful? Thanks so much for responding so promptly on a Sunday!

FalconBFR, 2021-08-15:
Would some screenshots/recordings / a call when you are free help? Thanks so much!

FalconBFR, 2021-08-15:
It seems like this issue is occurring with a new project that I just started too! it is not affecting old projects.

dhavalgujar, 2021-08-15:
> @dhavalgujar Thanks for the quick reply! I tried building it with idf.py in linux and the issue still persists. Thanks so much for responding so promptly on a Sunday!

You're welcome!
Regarding the build command that is failing, you seem to be on the master branch.
Are you wanting to use that branch, or `release/beta`?
Can you help us with the commit ID / version of IDF that you are using?
Also, is there a specific example that you are trying to build?

> Would some screenshots/recordings / a call when you are free help? Thanks so much!

Sure, if we can't figure it out this way, let's do that.

> It seems like this issue is occurring with a new project that I just started too! it is not affecting old projects.

In that case, I think you need to check that you are using [release/v4.3](https://github.com/espressif/esp-idf/tree/release/v4.3) of IDF, and then redo the steps given [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#get-started-set-up-tools) once.

FalconBFR, 2021-08-15:
Thanks for your help. 
Regarding the idf version, I am using `ESP-IDF v4.3-dirty`
Regarding the branch, which do you recommend that I use? Thanks!

FalconBFR, 2021-08-15:
With the Linux, it is actually a fresh install! (No esp-idf was ever installed on that computer until today)

dhavalgujar, 2021-08-15:
> Regarding the idf version, I am using `ESP-IDF v4.3-dirty`

I see, this seems okay.
Although, if there are changes that you have made to IDF which are not needed, you can reset them.

> Regarding the branch, which do you recommend that I use? Thanks!

The master branch is stable and tested, but works with an older version of AWS IoT.
`release/beta` uses the latest version (202103.00) of AWS IoT but the beta branch might undergo few changes in coming weeks and months, but all the examples are functional.

> With the Linux, it is actually a fresh install! (No esp-idf was ever installed on that computer until today)

Oh!
Have you followed all steps from [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#setting-up-development-environment)?
Are you able to compile other examples within IDF itself, `hello_world`, perhaps?

FalconBFR, 2021-08-15:
@dhavalgujar Thanks for your help. I will try the beta branch first and I will get back to you on whether it works or not! Thanks so much!

Regarding your questions:
I followed all the steps from 'here' during installation
I am able to compile other sample projects / my existing projects. 
I also experienced this 'improperly configured' error in another project (smartconfig) but I was able to resolve it by making some changes to the CMakeLists.txt

Something completely unrelated:
Another example that I am trying is 'SmartConfig'. With that example, it ruined perfectly. However, once I copied the code to my existing project, there are some errors with some modules not existing etc (esp_wpa2.h and nvs_flash.h). It is the same as another person on YouTube: [YouTube video 17:43](https://youtu.be/v81DcGf6Slg?t=1043) but that is an entirely different issue.

Thanks so much!