espressif/esp-aws-iot #69: Missing CMakeLists.txt file on release/beta branch.

rubengarridocecotec, 2021-09-30:
Hi!

I'm trying to add esp-aws-iot as a component on esp-idf sdk but I'm getting a CMake configure error because esp-idf tool can not compile esp-aws-iot component since CMakeLists.txt file is missing.

I only get this issue on release/beta branch, master branch contains CMakeList.txt file and I can compile the project without any issue. This is the error log I got:

`esp-idf/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added`

SolidStateLEDLighting, 2021-11-06:
rubengarridocecotec -- If you find a way forward on this -- can you reach out to me with a the answer?

jobo89, 2022-08-31:
It looks like the CMakeLists.txt file is not in the latest release. You can try:
```
cd ~/[path]/esp-idf/components
git clone -b "release/v3.1.x" --recursive https://github.com/espressif/esp-aws-iot
```

imdahisaria, 2025-03-18:
Can anybody provide a Cmakelist.txt for the esp-aws-iot folder ?