espressif/esp-aws-iot #74: Beta ?? Fleet Provisioning example??

SolidStateLEDLighting, 2021-11-15:
I'm digging through the Beta -- can I safely assume that the mqtt example is going to lead to success in Fleet Provisioning??

I already achieved Fleet Provisioning in the old esp-aws-iot software, but the new build is a whole new animal and I'm starting over with Fleet Provisioning now.

Fleet Provisioning (and possibly the option for Certificate rotation) is a must have for all large scale developers.    Why would you pass this over when your largest customers would most certainly require it???

SolidStateLEDLighting, 2021-11-24:
I have completed Fleet Provision by claim.  I used the mqtt tls mutual authentication example as my guide and everything works well.   The only required change is that the user must increase the MQTT_NETWORK_BUFFER_SIZE to 4096.

SolidStateLEDLighting, 2021-12-02:
BIG WARNING:   Cert/Keys which are received from AWS (for Provisioning) must be cleared of all string "/n" and those replaced with '/n' characters at the end of each 64 char block.  Be extra careful that final string retains '/n' and '/0' on the end or cert/key may fail.   You may need to append stored cert/key with '\0' character after pulling them back out of NVM so they totally match the format of any normally embedded cert/key.

RobertFoll, 2021-12-29:
Are there any guides or examples on using fleet provisioning with ESP32?

SolidStateLEDLighting, 2021-12-29:
I made Fleet Provisioning work with the older (current) sdk and also the new beta library.

It is not much more than a series of subscriptions and publishing.   All of it is pretty well explained in the AWS IOT Core documents.

Order of operations is critical to success.

The bear is going to be getting all your roles and policies, and template all set up correctly at AWS.

One last thing that you're not going to like is that when you finally get your cert and key -- you'll will likely need to strip out the 2-character LFs  "\n"(which are ascii printable as \n) and replace those with the non-printable single character '\n'  (you can see them in action but they are not characters on the screen) -- before you store your cert/key to NVM.

Once you have the cert/key then you restart your log-in process as demonstrated for Shadow work.

The need that is supremely over-looked right now is the full integration of all the features in one large example.   Fleet/Shadow/Jobs/OTA     They have OTA details abstracted away in a form that is completely disassociated from the fundamental Fleet/Shadow process.   I'm working on the Jobs/OTA portion of it right now.

Keith

RobertFoll, 2021-12-29:
Thank you very much for your detailed response.  It is very much appreciated!  I will try this out when I have time!

SolidStateLEDLighting, 2021-12-29:
I forget to mention one important thing..  I write in C++.   That adds a bit more difficulty in some situations because of name mangling and stricter structure initialization.    Some of what I wrote previously you may not see if writing in just C.

SolidStateLEDLighting, 2022-02-03:
I have fleet provisioning working now....

closed 2022-02-03 by SolidStateLEDLighting