espressif/esp-aws-iot #75: Beta ?? Master Sample Project???   We need it.

SolidStateLEDLighting, 2021-11-18:
I do appreciate the idea that you have isolated all the sub-projects -- but in the real world we need many of them all together.

Anyone who is making a serious product would use all of these together:

Fleet Provisioning
Thing_Shadow
Jobs
OTA

I'm looking at the need to merge all of the build files and that means I'm burning up time on things that I don't normally handle.

I'm having linking errors -- and I'm faced with challenge of figuring out how to merge the some of the kconfig_menu.json.

Can we get a master example project with all services merged??    Testing everything together is the final challenge.

SolidStateLEDLighting, 2021-11-23:
I have this master project with all sub-module together and compiling/linking now.

shahpiyushv, 2021-11-23:
@SolidStateLEDLighting , good to know that you have a master project working. 

Meanwhile, we started off adding support for the [demos](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/202103.00/demos) that the original aws iot repository had, starting from release [202103.00](https://github.com/aws/aws-iot-device-sdk-embedded-C/releases/tag/202103.00). It took us quite some time since the structure had changed drastically and there were multiple different porting codes available and it was hard to choose and make it work. Nevertheless, a lot of progress has already been made. We will soon update to the latest release [202108.00](https://github.com/aws/aws-iot-device-sdk-embedded-C/releases/tag/202108.00) which has the fleet provisioning.

We also understand that a master example would be useful, however our priority has been to bring our examples at par with what AWS has in their repo. Once that is done, we will work on a master example which would be a combination of other examples.

SolidStateLEDLighting, 2021-11-23:
I should have my Fleet Provision working again in a day or two.    I had it working in the old sdk before so I know I have the AWS side all setup correctly.

I think the new software is working very well at this point.     I'm really hoping that you have OTA nailed down nicely.   We need that.

SolidStateLEDLighting, 2022-02-03:
I have all working together now, Shadow, Fleet Provisioning, OTA, etc.

closed 2022-02-03 by SolidStateLEDLighting