espressif/esp-aws-iot #76: aws_iot: failed! mbedtls_ssl_handshake returned -0x6800 (IDFGH-6259)

MaazSk, 2021-11-19:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         [Make|CMake|idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
-Operating System: Linux
-Using an IDE : VS Code.
-Power Supply: USB

## Problem Description
When I am trying to connect to AWS IOT server I am getting errors for a handshake as below

```aws_iot: failed! mbedtls_ssl_handshake returned -0x6800```. I get this error when I am using the GSM module to connect the AWS server. But when I connect using WiFi I don't get these errors and it connects smoothly.

The logs which I receive are as follows:

> I (23399) aws_iot: Shadow Connect
> I (26109) mbedtls: ssl_tls.c:8209 => handshake
> 
> I (26109) mbedtls: ssl_cli.c:3787 client state: 0
> 
> I (26109) mbedtls: ssl_tls.c:2850 => flush output
> 
> I (26109) mbedtls: ssl_tls.c:2862 <= flush output
> 
> I (26119) mbedtls: ssl_cli.c:3787 client state: 1
> 
> I (26129) mbedtls: ssl_tls.c:2850 => flush output
> 
> I (26129) mbedtls: ssl_tls.c:2862 <= flush output
> 
> I (26139) mbedtls: ssl_cli.c:805 => write client hello
> 
> I (26149) mbedtls: ssl_tls.c:3289 => write handshake message
> 
> I (26149) mbedtls: ssl_tls.c:3448 => write record
> 
> I (26159) mbedtls: ssl_tls.c:2850 => flush output
> 
> I (26159) mbedtls: ssl_tls.c:2869 message length: 275, out_left: 275
> 
> I (26169) mbedtls: ssl_tls.c:2874 ssl->f_send() returned 275 (-0xfffffeed)
> 
> I (26179) mbedtls: ssl_tls.c:2902 <= flush output
> 
> I (26179) mbedtls: ssl_tls.c:3581 <= write record
> 
> I (26189) mbedtls: ssl_tls.c:3425 <= write handshake message
> 
> I (26189) mbedtls: ssl_cli.c:1255 <= write client hello
> 
> I (26199) mbedtls: ssl_cli.c:3787 client state: 2
> 
> I (26199) mbedtls: ssl_tls.c:2850 => flush output
> 
> I (26209) mbedtls: ssl_tls.c:2862 <= flush output
> 
> I (26219) mbedtls: ssl_cli.c:1684 => parse server hello
> 
> I (26219) mbedtls: ssl_tls.c:4422 => read record
> 
> I (26229) mbedtls: ssl_tls.c:2631 => fetch input
> 
> I (26229) mbedtls: ssl_tls.c:2792 in_left: 0, nb_want: 5
> 
> I (31239) mbedtls: ssl_tls.c:2816 in_left: 0, nb_want: 5
> 
> I (31239) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned -26624 (-0x6800)
> 
> W (31239) mbedtls: ssl_tls.c:5084 mbedtls_ssl_fetch_input() returned -26624 (-0x6800)
> 
> W (31249) mbedtls: ssl_tls.c:4455 ssl_get_next_record() returned -26624 (-0x6800)
> 
> W (31259) mbedtls: ssl_cli.c:1691 mbedtls_ssl_read_record() returned -26624 (-0x6800)
> 
> I (31269) mbedtls: ssl_tls.c:8219 <= handshake
> 
> E (31269) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
> I (31279) mbedtls: ssl_tls.c:8854 => write close notify
> 
> I (31279) mbedtls: ssl_tls.c:8870 <= write close notify
> 
> I (31289) mbedtls: ssl_tls.c:9063 => free
> 
> I (31289) mbedtls: ssl_tls.c:9128 <= free
> 
> E (31299) aws_iot: Shadow Connection Error

shahpiyushv, 2021-11-19:
@MaazSk , the various timeouts can be found here: https://github.com/espressif/esp-aws-iot/blob/249c573cbdcf653c03a6bf3a30f658f005856880/port/network_mbedtls_wrapper.c#L42

However, are you sure that the Internet connectivity on 4G is already sorted?

MaazSk, 2021-11-20:
> However, are you sure that the Internet connectivity on 4G is already sorted?

My internet connectivity is of 2G,  I am using a 2G GSM module.

closed 2022-01-06 by MaazSk