espressif/esp-aws-iot #79: subscribe_publish example not working

listout, 2021-11-28:
I've checked my certificates and keys, and everything works with a simple Arduino sketch. But I'm getting `E (15513) aws_iot: Failed ! mbedtls_ssl_read returned -0x7880` error when trying the subscribe_publish example. What am I doing wrong?

mbedTLS debug log:

```
I (4373) mbedtls: ssl_tls.c:8209 => handshake
I (4373) mbedtls: ssl_cli.c:3787 client state: 0
I (4383) mbedtls: ssl_tls.c:2850 => flush output
I (4383) mbedtls: ssl_tls.c:2862 <= flush output
I (4393) mbedtls: ssl_cli.c:3787 client state: 1
I (4393) mbedtls: ssl_tls.c:2850 => flush output
I (4403) mbedtls: ssl_tls.c:2862 <= flush output
I (4403) mbedtls: ssl_cli.c:805 => write client hello
I (4413) mbedtls: ssl_tls.c:3289 => write handshake message
I (4423) mbedtls: ssl_tls.c:3448 => write record
I (4423) mbedtls: ssl_tls.c:2850 => flush output
I (4433) mbedtls: ssl_tls.c:2869 message length: 254, out_left: 254
I (4443) mbedtls: ssl_tls.c:2874 ssl->f_send() returned 254 (-0xffffff02)
I (4443) mbedtls: ssl_tls.c:2902 <= flush output
I (4453) mbedtls: ssl_tls.c:3581 <= write record
I (4453) mbedtls: ssl_tls.c:3425 <= write handshake message
I (4463) mbedtls: ssl_cli.c:1255 <= write client hello
I (4473) mbedtls: ssl_cli.c:3787 client state: 2
I (4473) mbedtls: ssl_tls.c:2850 => flush output
I (4483) mbedtls: ssl_tls.c:2862 <= flush output
I (4483) mbedtls: ssl_cli.c:1684 => parse server hello
I (4493) mbedtls: ssl_tls.c:4422 => read record
I (4493) mbedtls: ssl_tls.c:2631 => fetch input
I (4503) mbedtls: ssl_tls.c:2792 in_left: 0, nb_want: 5
I (4783) mbedtls: ssl_tls.c:2816 in_left: 0, nb_want: 5
I (4783) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (4793) mbedtls: ssl_tls.c:2837 <= fetch input
I (4793) mbedtls: ssl_tls.c:2631 => fetch input
I (4803) mbedtls: ssl_tls.c:2792 in_left: 5, nb_want: 96
I (4803) mbedtls: ssl_tls.c:2816 in_left: 5, nb_want: 96
I (4813) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned 91 (-0xffffffa5)
I (4823) mbedtls: ssl_tls.c:2837 <= fetch input
I (4823) mbedtls: ssl_tls.c:4496 <= read record
I (4833) mbedtls: ssl_cli.c:2000 server hello, total extension length: 15
I (4843) mbedtls: ssl_cli.c:2200 <= parse server hello
I (4843) mbedtls: ssl_cli.c:3787 client state: 3
I (4853) mbedtls: ssl_tls.c:2850 => flush output
I (4853) mbedtls: ssl_tls.c:2862 <= flush output
I (4863) mbedtls: ssl_tls.c:5766 => parse certificate
I (4863) mbedtls: ssl_tls.c:4422 => read record
I (4873) mbedtls: ssl_tls.c:2631 => fetch input
I (4873) mbedtls: ssl_tls.c:2792 in_left: 0, nb_want: 5
I (4883) mbedtls: ssl_tls.c:2816 in_left: 0, nb_want: 5
I (4893) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (4893) mbedtls: ssl_tls.c:2837 <= fetch input
I (4903) mbedtls: ssl_tls.c:2631 => fetch input
I (4913) mbedtls: ssl_tls.c:2792 in_left: 5, nb_want: 4996
I (4913) mbedtls: ssl_tls.c:2816 in_left: 5, nb_want: 4996
I (4923) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned 4991 (-0xffffec81)
I (4933) mbedtls: ssl_tls.c:2837 <= fetch input
I (4973) mbedtls: ssl_tls.c:4496 <= read record
I (5063) mbedtls: ssl_tls.c:5974 <= parse certificate
I (5063) mbedtls: ssl_cli.c:3787 client state: 4
I (5063) mbedtls: ssl_tls.c:2850 => flush output
I (5073) mbedtls: ssl_tls.c:2862 <= flush output
I (5073) mbedtls: ssl_cli.c:2568 => parse server key exchange
I (5083) mbedtls: ssl_tls.c:4422 => read record
I (5083) mbedtls: ssl_tls.c:2631 => fetch input
I (5093) mbedtls: ssl_tls.c:2792 in_left: 0, nb_want: 5
I (5103) mbedtls: ssl_tls.c:2816 in_left: 0, nb_want: 5
I (5103) mbedtls: ssl_tls.c:2817 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (5113) mbedtls: ssl_tls.c:2837 <= fetch input
I (5113) mbedtls: ssl_tls.c:2631 => fetch input
I (5123) mbedtls: ssl_tls.c:2792 in_left: 5, nb_want: 338
I (5133) mbedtls: ssl_tls.c:2816 in_left: 5, nb_want: 338
[... 183 lines cut ...]
I (8923) mbedtls: ssl_tls.c:2471 <= decrypt buf
I (8933) mbedtls: ssl_tls.c:5281 got an alert message, type: [1:0]
I (8933) mbedtls: ssl_tls.c:5296 is a close notify message
W (8943) mbedtls: ssl_tls.c:4480 mbedtls_ssl_handle_message_type() returned -30848 (-0x7880)
W (8953) mbedtls: ssl_tls.c:8460 mbedtls_ssl_read_record() returned -30848 (-0x7880)
E (8963) aws_iot: Failed
  ! mbedtls_ssl_read returned -0x7880
I (8973) mbedtls: ssl_tls.c:8854 => write close notify
I (8973) mbedtls: ssl_tls.c:5361 => send alert message
I (8983) mbedtls: ssl_tls.c:3448 => write record
I (8983) mbedtls: ssl_tls.c:1446 => encrypt buf
I (8993) mbedtls: ssl_tls.c:1780 <= encrypt buf
I (8993) mbedtls: ssl_tls.c:2850 => flush output
I (9003) mbedtls: ssl_tls.c:2869 message length: 31, out_left: 31
I (9013) mbedtls: ssl_tls.c:2874 ssl->f_send() returned 31 (-0xffffffe1)
I (9013) mbedtls: ssl_tls.c:2902 <= flush output
I (9023) mbedtls: ssl_tls.c:3581 <= write record
I (9023) mbedtls: ssl_tls.c:5374 <= send alert message
I (9033) mbedtls: ssl_tls.c:8870 <= write close notify
I (9043) mbedtls: ssl_tls.c:9063 => free
I (9043) mbedtls: ssl_tls.c:9128 <= free
E (9053) subpub: Error(-12) connecting to aac2p94uhlhhz-ats.iot.us-east-2.amazonaws.com:8883
```

Thanks

listout, 2021-11-28:
I was able to fix the issue by adding the following defines in my main .c file

```c
#define AWS_IOT_MQTT_CLIENT_ID         "aws_esp32_test" ///< MQTT client ID should be unique for every device
#define AWS_IOT_MY_THING_NAME          "aws_esp32_test" ///< Thing Name of the Shadow this device is associated with
```

It would be helpful for a newcomer like (and other) if the maintainers updated the readme about this. I'm sure there would be better method of solving the issue (like including an opting in menuconfig.

Thank you.

aleblsv, 2021-12-17:
I have the same issue with examples thing_shadow and subscribe_publish

@listout can you share your main.c file ?

shahpiyushv, 2021-12-18:
@listout , client id and thing name are indeed part of Kconfig option with some defaults. The default for client id is sufficient if your AWS IoT policy isn't strict. The README in example also mentions about changing the client id [here](https://github.com/espressif/esp-aws-iot/tree/master/examples#optional-set-client-id) if required. The Thing shadow example also mentions about thing name [here](https://github.com/espressif/esp-aws-iot/tree/master/examples/thing_shadow#set-thing-name). Can you elaborate what information you could not find in the READMEs?

aleblsv, 2021-12-19:
@shahpiyushv Can you please verify examples are working?

shahpiyushv, 2021-12-20:
@aleblsv , just tried the subscribe_publish example on esp32/esp32c3 with idf 4.3.1 and it is working fine for me.

```
I (776) wifi:mode : sta (3c:71:bf:ce:03:c0)
I (776) wifi:enable tsf
I (776) subpub: AWS IoT SDK Version 3.0.1-
I (1396) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2116) wifi:state: init -> auth (b0)
I (2156) wifi:state: auth -> assoc (0)
I (2166) wifi:state: assoc -> run (10)
I (2256) wifi:connected with xxxxxxxx, aid = 2, channel 6, BW20, bssid = 6e:41:99:eb:ad:e1
I (2256) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (2256) wifi:pm start, type: 1

I (2596) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (3566) esp_netif_handlers: sta ip: 192.168.74.253, mask: 255.255.255.0, gw: 192.168.74.71
I (3566) subpub: Connecting to AWS...
W (4586) wifi:<ba-add>idx:0 (ifx:0, 6e:41:99:eb:ad:e1), tid:0, ssn:3, winSize:64
I (8856) subpub: Subscribing...
I (9466) subpub: Stack remaining for task 'aws_iot_task' is 3588 bytes
I (11216) subpub: Subscribe callback
I (11216) subpub: test_topic/esp32	hello from ESP32 (QOS0) : 0 
I (11816) subpub: Subscribe callback
I (11816) subpub: test_topic/esp32	hello from ESP32 (QOS1) : 1 
I (11916) subpub: Stack remaining for task 'aws_iot_task' is 3588 bytes
I (13256) subpub: Subscribe callback
I (13256) subpub: test_topic/esp32	hello from ESP32 (QOS0) : 2 
I (13866) subpub: Subscribe callback
I (13866) subpub: test_topic/esp32	hello from ESP32 (QOS1) : 3 
I (13976) subpub: Stack remaining for task 'aws_iot_task' is 3588 bytes
I (15606) subpub: Subscribe callback
I (15616) subpub: test_topic/esp32	hello from ESP32 (QOS0) : 4 
I (16226) subpub: Subscribe callback
I (16226) subpub: test_topic/esp32	hello from ESP32 (QOS1) : 5 
I (16326) subpub: Stack remaining for task 'aws_iot_task' is 3588 bytes
I (17656) subpub: Subscribe callback
I (17656) subpub: test_topic/esp32	hello from ESP32 (QOS0) : 6 
I (18276) subpub: Subscribe callback
I (18276) subpub: test_topic/esp32	hello from ESP32 (QOS1) : 7 
I (18376) subpub: Stack remaining for task 'aws_iot_task' is 3588 bytes
```

 Please remember to set the following

- main/certs/certificate.pem.crt file
- main/certs/private.pem.key  file
- CONFIG_WIFI_SSID
- CONFIG_WIFI_PASSWORD
- CONFIG_AWS_EXAMPLE_CLIENT_ID
- CONFIG_AWS_IOT_MQTT_HOST

No other changes should be required if you AWS side configuration is also done correctly.

aleblsv, 2021-12-20:
@shahpiyushv can you share your credentials certificate.pem.crt, private.pem.key, CONFIG_AWS_EXAMPLE_CLIENT_ID and CONFIG_AWS_IOT_MQTT_HOST ?
Also can you attach the policy ?
Thanks

aleblsv, 2021-12-20:
@shahpiyushv I noticed missing line in my log (wifi:enable tsf). 
So, I have clean all configuration with idf.py set-target esp32
I'm using ESP32-WROVER-E
Now I see the wifi:enable tsf and subscribe_publish example works as expected
Thanks!

I (665) subpub: Setting WiFi configuration SSID xxxx...
I (665) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (775) wifi:mode : sta (10:52:1c:81:9c:d8)
I (775) wifi:enable tsf
I (775) subpub: AWS IoT SDK Version 3.0.1-
I (1745) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
I (2475) wifi:state: init -> auth (b0)
I (2485) wifi:state: auth -> init (1c0)
I (2485) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (4535) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (4535) wifi:state: init -> auth (b0)
I (4545) wifi:state: auth -> assoc (0)
I (4555) wifi:state: assoc -> run (10)
I (4565) wifi:connected with xxxx, aid = 4, channel 9, BW20, bssid = e6:38:08:0b:5e:bf
I (4565) wifi:security: WPA2-PSK, phy: bgn, rssi: -59
I (4575) wifi:pm start, type: 1
I (4625) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5565) esp_netif_handlers: sta ip: 192.168.181.115, mask: 255.255.255.0, gw: 192.168.181.183
I (5935) subpub: Connecting to AWS...
W (6465) wifi:<ba-add>idx:0 (ifx:0, e6:38:08:0b:5e:bf), tid:0, ssn:3, winSize:64
W (6775) wifi:<ba-add>idx:1 (ifx:0, e6:38:08:0b:5e:bf), tid:5, ssn:0, winSize:64
W (7185) wifi:<ba-add>idx:2 (ifx:0, e6:38:08:0b:5e:bf), tid:4, ssn:0, winSize:64
I (9855) subpub: Subscribing...
I (10355) subpub: Stack remaining for task 'aws_iot_task' is 12812 bytes
I (11995) subpub: Stack remaining for task 'aws_iot_task' is 12812 bytes