espressif/esp-aws-iot #81: OTA update failing due to signature verification

shaikhzahid06, 2021-12-03:
Hi,
I have completed aws prerequisites for OTA job and also generated and flashed the code signing certificate. The OTA agent receives the update from the job but fails after receiving the last block due to Signature verification. Below is the console output.
 (313241) OTA: Received valid file block: Block index=223, Size=2544
I (313251) OTA: Received final block of the update.
I (313251) PKCS11: Initializing NVS partition: "storage"
I (313271) PKCS11: failed nvs open 4354
I (313271) OTA: No such certificate file: Code Verify Key. Using certificate in ota_config.h.
I (314771) DEMO:  Received: 88   Queued: 88   Processed: 87   Dropped: 0
E (314921) OTA: Signature verification failed
I (316461) DEMO:  Received: 88   Queued: 88   Processed: 87   Dropped: 0
E (317611) OTA: Failed to close the OTA file: Error=(OtaPalSignatureCheckFailed:0x000000)
E (317611) OTA: Failed to ingest data block, rejecting image: ingestDataBlock returned error: OtaErr_t=-2
I (317621) OTA: otaPal_SetPlatformImageState, 3
W (317631) OTA: Set image as invalid!
I (317631) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
W (317641) esp_ota_ops: otadata partition is invalid, factory/ota_0 is boot partition
E (317651) OTA: Currently executing firmware not marked as valid, abort

andrew-elder, 2022-04-15:
@shaikhzahid06 - did you figure this out. I'm seeing the same thing.

loganbenda, 2023-01-09:
@shaikhzahid06 @andrew-elder I'm having the same issue, but haven't been able to solve yet.

`
I (187760) AWS_OTA: Received final block of the update.
I (187760) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
I (188770) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
E (188920) AWS_OTA: Failed to close the OTA file
I (189830) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
I (191110) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
I (192210) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
I (193510) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
I (194640) AWS_OTA:  Received: 282   Queued: 282   Processed: 281   Dropped: 0
E (194820) AWS_OTA: Failed to close the OTA file: Error=(OtaPalSignatureCheckFailed:0xe3000000)
E (194820) AWS_OTA: Failed to ingest data block, rejecting image: ingestDataBlock returned error: OtaErr_t=-2
I (194830) AWS_OTA: otaPal_SetPlatformImageState, 3
W (194830) AWS_OTA: Set image as invalid!
I (194840) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (194840) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0x3/0x0
I (194850) esp_ota_ops: aws_esp_ota_set_boot_flags: 3 0
I (194860) esp_ota_ops: [1] aflags/seq:0x3/0x0, pflags/seq:0x2/0x1
I (195640) AWS_OTA:  Received: 282   Queued: 282   Processed: 281  
`

When going through the README this section didn't make sense to me. Seems like it should be the path to the generated aws_codesign.crt.

4. Now, to create an OTA update job, using the AWS IoT console, follow the steps mentioned [here](https://docs.aws.amazon.com/freertos/latest/userguide/ota-console-workflow.html).  
For **"Path name of code signing certificate on device"**, put the following value: 
```
Code Verify Key
```
This corresponds to `pkcs11configLABEL_CODE_VERIFICATION_KEY` in the `core_pkcs11_config.h` file.