espressif/esp-aws-iot #83: Excessive d_ram usage on OTA example.

AlexRadek, 2021-12-28:
Hey everyone, I noticed that there is a massive usage of d_ram on the OTA mqtt example. I am using an ESP32.

Here is the "idf.py size" log when: 
static void setOtaInterfaces(OtaInterfaces_t* pOtaInterfaces)
{
...
pOtaInterfaces->mqtt.subscribe = mqttSubscribe;
...
}
**LOG:**
Total sizes:
 DRAM .data size:   17720 bytes
 DRAM .bss  size:   82552 bytes
Used static DRAM:  100272 bytes (  80464 available, 55.5% used)
Used static IRAM:   89194 bytes (  41878 available, 68.0% used)
      Flash code:  615455 bytes
    Flash rodata:  150968 bytes
Total image size:~ 873337 bytes (.bin may be padded larger)

Here is the "idf.py size" log when: 
static void setOtaInterfaces(OtaInterfaces_t* pOtaInterfaces)
{
...
pOtaInterfaces->mqtt.subscribe = NULL; 
...
}

**LOG**
Total sizes:
 DRAM .data size:   17720 bytes
 DRAM .bss  size:   26192 bytes
Used static DRAM:   43912 bytes ( 136824 available, 24.3% used)
Used static IRAM:   89194 bytes (  41878 available, 68.0% used)
      Flash code:  614187 bytes
    Flash rodata:  149928 bytes
Total image size:~ 871029 bytes (.bin may be padded larger)

Notice that the subscribe function is using 56430 bytes of ram! I believe there might be some kind of memory leak somewhere. I have tried tracing this and found that it happens whenever "callbackRecordList" in the "mqtt_subscription_manager.c" is being written in the function "SubscriptionManager_RegisterCallback" this jump in memory usage happens.

I am trying to integrate this OTA code into our system. However, it is impossible without some kind of optimization. Any help would be very much appreciated.

AlexRadek, 2022-01-03:
For those interested I have found the problem.

The amount of buffers is kind of big, thats why it uses so much d_ram.

You can change the amount by editing this variable: otaconfigMAX_NUM_OTA_DATA_BUFFERS.

closed 2022-01-03 by AlexRadek

davidallenmann, 2022-11-04:
Ran into the same thing. If you have a module with external RAM, you can also use the external RAM to hold arrays that use a lot of DRAM. 

1. Declare as pointers.
2. Allocate external PSRAM with heaps_caps_malloc
```
//static uint8_t otaNetworkBuffer[OTA_NETWORK_BUFFER_SIZE];
static uint8_t *otaNetworkBuffer;
// uint8_t decodeMem[otaconfigFILE_BLOCK_SIZE];
uint8_t *decodeMem;
//static OtaEventData_t eventBuffer[otaconfigMAX_NUM_OTA_DATA_BUFFERS];
static OtaEventData_t *eventBuffer;
```
```
    // allocate memory for OTA from external RAM
    eventBuffer = (OtaEventData_t *)heap_caps_malloc(otaconfigMAX_NUM_OTA_DATA_BUFFERS, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    otaNetworkBuffer = (uint8_t *)heap_caps_malloc(OTA_NETWORK_BUFFER_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    decodeMem = (uint8_t *)heap_caps_malloc(otaconfigFILE_BLOCK_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
```