espressif/esp-aws-iot #87: ESP-AWS-IoT support for  fleet provisioning

helaagardabbou, 2022-01-11:
Hello please when the ESP-AWS-IoT component will support fleet provisioning ?

SolidStateLEDLighting, 2022-01-13:
You can achieve Fleet Provisioning right now.   I did it both the older version and in this new beta version.   Everything is just a matter of subscription and publishing.   There is quite a bit to set up over at AWS, but the instructions there are correct and complete.   Order of operation is important.   And you'll have one sticking point when you receive the new certs.    AWS will send the certs to you with 2 character line feeds.   You'll need to strip those out and insert one character line feeds at the correct 64 chars positions.   Other than this --- the provisioning does work fairly well.

EmbeddedDBL, 2022-01-17:
Hello @SolidStateLEDLighting , could you explain a little more how to carry out the provisioning of fleets with OTA functionality of the release/beta version?

SolidStateLEDLighting, 2022-01-17:
All of the IOT services are not much more than subscriptions, publishing, and the handling of response messages.  

If you get your project organized correctly, you can do a client log-in (no Thing identity), a Fleet Provisioning, a Shadow connection (Thing identity), a Job service, and OTA -- with just pub/sub and messages.

The big problem with the supplied OTA project is that it is not made for integration with all the other services.   I deconstructed their process and merged it with all my other work.

I'm not using HTTPs for downloading binaries right now -- I do everything with MQTT.    MQTT is mandatory for control messages, but you can stream with it for the OTA binary downloads too.   So, I find it easier to just MQTT_ProcessLoop() for everything in one location.

The OTA Agent software example provided is written to be POSIX universal for a hand-full of hardware platforms.  So, it is very unfriendly to ESP32 directly.   And this guy (or girl) abstracts the work with function points (which drives me bananas),  because you can't follow the code in a progressive and logical way.   I like project organization that teaches you what to do (or reminds you what you did).

EmbeddedDBL, 2022-01-18:
@SolidStateLEDLighting Thanks for your feedback. I think I now have a clear understanding of the process. Reviewing the AWS documentation suggests the following workflow.

This is a fleet provisioning by claim ( JITP )
[image]

I will be working on conditioning the OTA example for fleet provisioning and I would appreciate if you know any (documentation or examples) that can support me.

And another question that does not let me sleep -_-
Is it possible to update the code signature with which the firmware file to be updated is signed via OTA, since this signature is in flash and has an expiration date in AWS?

SolidStateLEDLighting, 2022-01-19:
I think that Fleet Provision by Claim and JITP are slightly different things.

The OTA example has nothing to do with provisioning.   It is one of the reasons that I don't like it.  It does not get along with all other services.

I'm not completely up to speed on the code signing for OTA.   I make a custom OTA job and supply the cert in the window provided (third selection).   That cert is sent by AWS in the Job description.   This allows me for now to download a non-encrypted binary (AWS doesn't know that its unsigned).   You wouldn't be able to see any of this inside the OTA tool that they provide because all that is hidden behind the scenes.    I don't even know if you have the option of using my approach.   That OTA example is not documented well enough inside the code to know all that you need.   Externally -- no documentation at all.   As I mentioned in previous email -- I tore it apart and followed its pattern to achieve the similar results.

I believe that updating the code signature is something that can be achieved inside AWS Certificate Manager -- but I'm not using that service for OTA right now.

[image]<https://user-images.githubusercontent.com/94095787/150033039-acdab31c-ff33-4c21-b8ec-75320cc49037.png>

EmbeddedDBL, 2022-01-26:
@SolidStateLEDLighting 
Hello again.
I appreciate your comments.

Ask you if you used these AWS APIs for fleet provisioning by claim ? 
https://docs.aws.amazon.com/es_es/iot/latest/developerguide/fleet-provision-api.html#create-cert-csr-request-payload

This method called CSR is the same as by claim?

SolidStateLEDLighting, 2022-01-26:
Yes, I did fleet provisioning by claim.  I believe it is somewhat different than the process which uses the Certificate Signing Request.   I did not read up on the CSR method very much.

I did not use APIs.    There is no programming interface other than subscribing and publishing to topics which apply to the provisioning process (by claim).

You need to set up the AWS side with a provisioning template -- and all the correct roles and policies.   That takes careful attention to detail.  The AWS documents that explain all this are correct.

Keith

https://docs.aws.amazon.com/es_es/iot/latest/developerguide/fleet-provision-api.html#create-cert-csr-request-payload

EmbeddedDBL, 2022-02-01:
Thanks for your comment.
@SolidStateLEDLighting 
I now understand better what needs to be done, I think I am following the steps correctly but even so I am not getting a response from AWS.

1 Connect to MQTT with claim certificate
2 I subscribe to the answer topics $aws/certificates/create/payload-format/accepted , ....rejected
3 Post a message to the topic $aws/certificates/create/payload-format
This is where the question arises what message I have to publish since it mentions that it is an empty payload-format. 

I'm not sure if the problem is the message you sent when making the request. Thank you again for your comments.
Now try the CreateCertificateFromCsr method.

SolidStateLEDLighting, 2022-02-02:
Notes:

  1.  I have no idea how the provisioning with CSR works.   I didn't do it this way so I can't help you in this method.   I did it "by claim"
  2.   Yes -- read the docs carefully on all the topics needed for your specific action
  3.    {}   <--  (this is the empty payload, but literally nothing may also work)    I experimented and read documents to figure this out.

helaagardabbou, 2022-02-02:
Hello Try to subscribe to the answer topics $aws/certificates/create/json/accepted and publish an empty msg to  $aws/certificates/create/json

SolidStateLEDLighting, 2022-02-02:
Why would I want to follow this suggestion?   For what purpose?

helaagardabbou, 2022-02-02:
because payload format musht be cbor or json

SolidStateLEDLighting, 2022-02-02:
I have had success with Fleet Provisioning by Claim.

Why are you providing me this suggestion?

helaagardabbou, 2022-02-02:
This suggestion is a response for empty payload-format error of EmbeddedDBL

SolidStateLEDLighting, 2022-02-02:
Of course.... sorry for the confusion.   I'm just looking at this in my email utility.

helaagardabbou, 2022-02-02:
no problem welcome

EmbeddedDBL, 2022-02-02:
@helaagardabbou  That's right, first I subscribe to the answer topics $aws/certificates/create/cbor/accepted and rejected

Then I publish the empty message in cbor $aws/certificates/create/cbor  format but I do not get a response from the server, however when I send a badly formatted message I do get a response in the rejected response topic and this has me confused.

helaagardabbou, 2022-02-02:
@EmbeddedDBL the publish and the subscribe are done with the same client ? that's mean  You did subscribe in the device and the publish action with a console client?

SolidStateLEDLighting, 2022-02-02:
Please don't use cbor (compact binary object representation) until you need to stream data.    cbor can be used, but you have encode it with a cbor encoder.  The results you get have to be decoded before inspection.

Use json (javascript object notation) so you can easily read the message in when they come back to you.

helaagardabbou, 2022-02-02:
@SolidStateLEDLighting  I totally agree with you also u can easily know the error with json format

EmbeddedDBL, 2022-02-02:
@helaagardabbou If it is the same client with which I establish an MQTT connection, when I publish the request message in the IoT security/Certificates console I can see that a new certificate is created but I do not receive anything in the /accepted topic.

Ok I appreciate your comments I will change to Json

helaagardabbou, 2022-02-02:
@EmbeddedDBL  u must publish mqtt messgae with the device which is subscribed on the /accepted topic

SolidStateLEDLighting, 2022-02-02:
Everything is done in the same MQTT connection if that connection is authorized (policy/roles) are defined correctly at AWS.     If the cert/key are being created, that you are at the first step.  Make double sure you are subscribing correctly to the right topics.   Cloud watch can assist you in seeing the actions and there may be some help in seeing errors there (like what topics you have subscribed to, etc).

K.

EmbeddedDBL, 2022-02-02:
I publish this in the topic $aws/certificates/create/json in JSON format char payloadBuffer2[] = " "; with the intention that it fails to see if I choose an answer in the error topic.

[image]

When I post a correct message I get no response in my mqtt callback function char payloadBuffer2[] = "{\"certificateSigningRequest\":\" \" }";
[image]

SolidStateLEDLighting, 2022-02-02:
Here is what the topics look like for Fleet Provisioning by Claim...

// First Step
std::string create_topic = "$aws/certificates/create/json";                                                // This is a publish topic
std::string create_json_accepted_topic = "$aws/certificates/create/json/accepted";  // This is a subscribe topic
std::string create_json_rejected_topic = "$aws/certificates/create/json/rejected";     // This is a subscribe topic

// Second Step
std::string provision_json_topic = "$aws/provisioning-templates/Provisioning_Template_V2/provision/json";                                       // This is a publish topic
std::string provision_json_accepted_topic = "$aws/provisioning-templates/Provisioning_Template_V2/provision/json/accepted";  // This is a subscribe topic
std::string provision_json_rejected_topic = "$aws/provisioning-templates/Provisioning_Template_V2/provision/json/rejected";     // This is a subscribe topic

Notice that I must provide a template which is accessible to my provisioning process.

[image]<https://user-images.githubusercontent.com/94095787/152217780-95090b36-8139-4f58-a9cc-70bb5dcf26d7.png>

[image]<https://user-images.githubusercontent.com/94095787/152218487-ee692567-dbda-4907-8568-e94f57a47ced.png>

EmbeddedDBL, 2022-02-03:
@SolidStateLEDLighting 
Upon completion of step one before proceeding to step two, the response from the "$aws/certificates/create/json/accepted" topic is received in a callback registered in MQTT_Init( );

Here my problem arises I do not receive any response I do not know if it is my implementation or my AWS configuration

SolidStateLEDLighting, 2022-02-03:
You'll need to set up Cloud watch to confirm that the AWS system created a Publish Out message.  That message will be your payload response.

Instructions on how to get Cloud watch to show you all the IOT traffic is documented.    Just follow the the documents.

Everything has to be perfect for everything to work.   Order of operation is critical.   Having your topics correct is critical.

SolidStateLEDLighting, 2022-02-03:
Here is another serious possibility....

The demos are not well written (but they are simple enough).  They block in the MQTT_ProcessLoop call for a small period.

The time it takes for you to receive a subscription or publish is small.

The time it takes for you to receive a payload is about 40 times longer in some cases.   So, you'll need to extend out your time in MQTT_ProcessLoop long enough to be sure you receive the payload.

I have written my project here to never leave MQTT_ProcessLoop area unless I have something to send.    Not leaving this loop is important because you don't know when the system is sending the MQTT keep alive pings -- and your code must process them behind the scenes for you in this same loop.

K.

EmbeddedDBL, 2022-02-10:
I already achieved provisioning by claim using CreateCertificateFromCsr it required more work but it works great.
I couldn't get a response with CreateKeysAndCertificate so I opted for CreateCertificateFromCsr.
Just thank you for your comments @SolidStateLEDLighting  and @helaagardabbou 👍

SolidStateLEDLighting, 2022-02-11:
Nice job!   Congratulations.

helaagardabbou, 2022-02-11:
Great! Congratulations !!! @EmbeddedDBL

SolidStateLEDLighting, 2022-05-24:
> It would be great if examples for fleet provisioning could be added. For those that have made it work, would it be possible to share some code?

I made everything work, but my project would be considered somewhat complex for "sample standards".

The key is to subscribe to the topics -- ask for the messages that you need -- publish at the correct times -- wrap all of it into one task that loops back around to MQTT_ProcessLoop(pMqttContext, 100) on a continuous basis.

The other side of the challenge is to get AWS set up correctly with roles and policies.   You must follow the AWS instructions to perfection in exactly the right order to get everything to work.

twlawrenceko, 2022-06-02:
I have tried to do fleet provisioning but seems like I am not receiving the certificate message by subscribing `$aws/certificates/create/json/accepted` topic. 
When I tried to do publish `$aws/certificates/create/json` in MQTT Test Client I am able to see the subscribed message showing up correctly in MQTT Test Client. However, when I tried to do it on ESP32 it does not show up in MQTT Test Client, however, the certificate is being generated as I can see it under AWS IoT Certificates webpage.

Here is my piece of code:
```
int32_t FleetProvisioning( void )
{
    /* Define Fleet Provisioning Subscribe topics */
    const char * sTopicFilter = "$aws/certificates/create/json/accepted";
    uint16_t sTopicFilterLength = 39;
    
    /* Define Fleet Provisioning Publish topics */
    const char * pTopicFilter = "$aws/certificates/create/json";
    int32_t pTopicFilterLength = 30;
    const char * pPayload = "{}";
    size_t payloadLength = 0;

    int returnStatus = EXIT_SUCCESS;
    MQTTStatus_t mqttStatus;
    MQTTContext_t * pMqttContext = &mqttContext;
    uint8_t publishIndex = MAX_OUTGOING_PUBLISHES;
    MQTTSubscribeInfo_t pSubscriptionList[ 1 ];

    assert( pMqttContext != NULL );
    assert( pTopicFilter != NULL );
    assert( pTopicFilterLength > 0 );
    assert( sTopicFilter != NULL );
    assert( sTopicFilterLength > 0 );

    /* Start with everything at 0. */
    ( void ) memset( ( void * ) pSubscriptionList, 0x00, sizeof( pSubscriptionList ) );

    /* This example subscribes to only one topic and uses QOS1. */
    pSubscriptionList[ 0 ].qos = MQTTQoS1;
    pSubscriptionList[ 0 ].pTopicFilter = sTopicFilter;
    pSubscriptionList[ 0 ].topicFilterLength = sTopicFilterLength;

    /* Generate packet identifier for the SUBSCRIBE packet. */
    globalSubscribePacketIdentifier = MQTT_GetPacketId( pMqttContext );

    /* Send SUBSCRIBE packet. */
    mqttStatus = MQTT_Subscribe( pMqttContext,
                                 pSubscriptionList,
                                 sizeof( pSubscriptionList ) / sizeof( MQTTSubscribeInfo_t ),
                                 globalSubscribePacketIdentifier );

    if( mqttStatus != MQTTSuccess )
    {
        LogError( ( "Failed to send SUBSCRIBE packet to broker with error = %u.",
                    mqttStatus ) );
        returnStatus = EXIT_FAILURE;
    }
    else
    {
        LogInfo( ( "SUBSCRIBE topic %.*s to broker.",
                   sTopicFilterLength,
                   sTopicFilter ) );
    }

    /* Get the next free index for the outgoing publish. All QoS1 outgoing
     * publishes are stored until a PUBACK is received. These messages are
     * stored for supporting a resend if a network connection is broken before
     * receiving a PUBACK. */
    returnStatus = getNextFreeIndexForOutgoingPublishes( &publishIndex );

    if( returnStatus == EXIT_FAILURE )
    {
        LogError( ( "Unable to find a free spot for outgoing PUBLISH message." ) );
    }
    else
[... 54 lines cut ...]

    

    return returnStatus;
}
```

And the log:
```
I (5608) coreMQTT: Establishing a TLS session to a3s5xxxxkgux4b-ats.iot.us-east-1.amazonaws.com:8883.
I (7638) coreMQTT: Packet received. ReceivedBytes=2.
I (7638) coreMQTT: CONNACK session present bit set.
I (7638) coreMQTT: Connection accepted.
I (7648) coreMQTT: Received MQTT CONNACK successfully from broker.
I (7658) coreMQTT: MQTT connection established with the broker.
I (7658) coreMQTT: MQTT connection successfully established with broker.
I (7668) coreMQTT: An MQTT session with broker is re-established. Resending unacked publishes.
I (7678) coreMQTT: SUBSCRIBE topic $aws/certificates/create/json/accepted to broker.
I (7688) coreMQTT: Published payload: {}
I (7688) coreMQTT: PUBLISH sent for topic $aws/certificates/create/json to broker with packet ID 2.
E (8288) esp-tls-mbedtls: read error :-0x004C:
E (8298) esp-tls-mbedtls: read error :-0x004C:
```

Any help would be appreciated.

SolidStateLEDLighting, 2022-06-02:
The full process requires that any time you do anything -- you enter the MQTT_ProcessLoop().

Your MQTT_Subscribe() return result confirms that you have sent the request, but only a return value in MQTT_ProcessLoop() will confirm that you have actually been subscribed.

So, a good sample that works it a bit more complicated than what you have written below.

Use cloud watch also to see what the system is sending you.

All your AWS roles, policies, and provision template must be meticulously accurate.

twlawrenceko, 2022-06-04:
@SolidStateLEDLighting  Thanks for the information.

I have tested with fleetprovisioning.py script to make sure fleet provisioning can be done. Now trying to convert onto ESP32.

I am following this [example](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c) provided by AWS embedded C SDK but found that their are [loading credentials into PKCS11 module](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/5da3a389cda566469ee5d79142961d30461ea570/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c#L508)

I am not quite familiar with PKCS, but is this a required procedure to do on ESP32 for fleet provisioning?

Thanks!

bakermat, 2022-06-04:
I had some time to look at this today. The part that got me is that I needed to increase the network buffers (used 5000), as I just was not receiving any responses to my subscribe to `$aws/certificates/create/json/accepted"` while I had it configured and saw CloudWatch publishing the data.

I'm now seeing the JSON response, the original question still stands though: while it's just some API calls that are made, it would be handy to have fleet provisioning baked in in the next version. Even now that I've got the certificate info, I still need to store it securely and add custom code to handle that part and avoid re-registering. Not the end of the world but there are clearly more people looking at this...

SolidStateLEDLighting, 2022-06-04:
Yes, the buffer by default is way too small for the certificate info....  I published this comment before but forgot to mention it in the last email.

I store the certs in NVS and look for them when I start up.   If I have them, I register as the MQTT client (with thing name).   If I don't have them, then I drop into provisioning mode (without thing name).

You'll notice that your certificate attachments are growing under your thing at AWS.   The docs recommend using a job to clear out old certs (security cert rotation).

But, that is a long way off because next you'll be wanting to integrate OTA (with jobs).

SolidStateLEDLighting, 2022-06-04:
I think the PKCS11 (public key cryptography standards) stuff is important ....   I just follow, when possible, their examples.

Their OTA sample is completely different from all the other examples and doesn't follow the normal pattern, but you can make it follow the pattern if you break it down.   It will integrate nicely when all the correct subscribes and publishes.

twlawrenceko, 2022-06-04:
I tried to follow the example AWS provided in their embedded C SDK, but having some library linkage issue.
```
../main/mqtt_operations.c:51:10: fatal error: mbedtls_pkcs11_posix.h: No such file or directory
 #include "mbedtls_pkcs11_posix.h"
```

Here is my CMakeLists.txt file, mostly followed what they have in their example.
```
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set( DEMO_NAME "fleet_provisioning_by_claim" )

set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common
						 "${CMAKE_CURRENT_LIST_DIR}/../../libraries/ota-for-aws-iot-embedded-sdk"
						 "${CMAKE_CURRENT_LIST_DIR}/../../libraries/coreMQTT"
						 "${CMAKE_CURRENT_LIST_DIR}/../../libraries/corePKCS11"
						 "${CMAKE_CURRENT_LIST_DIR}/../../libraries/backoffAlgorithm"
						 "${CMAKE_CURRENT_LIST_DIR}/../../libraries/common/posix_compat"
   )

# Include MQTT library's source and header path variables.
include( ${CMAKE_CURRENT_LIST_DIR}/../../libraries/coreMQTT/coreMQTT/mqttFilePaths.cmake )

# Include backoffAlgorithm library file path configuration.
include( ${CMAKE_CURRENT_LIST_DIR}/../../libraries/backoffAlgorithm/backoffAlgorithm/backoffAlgorithmFilePaths.cmake )

# Include Fleet Provisioning library's source and header path variables.
include(
    ${CMAKE_CURRENT_LIST_DIR}/../../libraries/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/fleetprovisioningFilePaths.cmake )

# Set path to corePKCS11 and it's third party libraries.
set(COREPKCS11_LOCATION "${CMAKE_CURRENT_LIST_DIR}/../../libraries/corePKCS11")
set(CORE_PKCS11_3RDPARTY_LOCATION "${COREPKCS11_LOCATION}/source/dependency/3rdparty")

# Include PKCS #11 library's source and header path variables.
include( ${COREPKCS11_LOCATION}/corePKCS11/pkcsFilePaths.cmake )

list(APPEND PKCS_SOURCES
    "${CORE_PKCS11_3RDPARTY_LOCATION}/mbedtls_utils/mbedtls_utils.c"
)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ota_mqtt)

target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/root_cert_auth.pem" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/client.crt" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/client.key" TEXT)
```

What's interesting is in the previous branch of `esp-aws-iot` they are using this [aws-iot-device-sdk-embedded-C](https://github.com/aws/aws-iot-device-sdk-embedded-C) repo as a base, but now they are completely using individual ones.

How did you manage to include all the necessary libraries for PKCS11?

SolidStateLEDLighting, 2022-06-04:
I just followed their examples -- much like what you have shown here in email.

This is part of my cmakelist.txt in the root directory:

# Set path to corePKCS11 and it's third party libraries.
set(COREPKCS11_LOCATION "${PROJECT_DIR}/aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11")
set(3RDPARTY_LOCATION "${PROJECT_DIR}/aws-iot-device-sdk-embedded-C/libraries/source/dependency/3rdparty")
set(CORE_PKCS11_3RDPARTY_LOCATION "${COREPKCS11_LOCATION}/source/dependency/3rdparty")

# Include PKCS #11 library's source and header path variables.
include( ${COREPKCS11_LOCATION}/pkcsFilePaths.cmake )

Notice where the corePKCS11 is based on the include references.

set(COMPONENT_SRCS        contains....
"${PKCS_SOURCES}"

set(COMPONENT_ADD_INCLUDEDIRS       contains....
"${CORE_PKCS11_3RDPARTY_LOCATION}/mbedtls_utils/mbedtls_utils.c"
"${CORE_PKCS11_3RDPARTY_LOCATION}/mbedtls_utils/mbedtls_error.c"

SolidStateLEDLighting, 2022-06-04:
Last posting was my cmake file inside the source file directory.

I have another cmake file in the root project directory which holds my project definition and my target_add_binary_data declarations.

You seem to have wrapped it up all in one file.

twlawrenceko, 2022-06-05:
Can you provide your CMakeLists.txt in the root directory and also for the project directory? I seem to have issue migrating into the root directory.

Thanks!

SolidStateLEDLighting, 2022-06-05:
My project contains client, MQTT, provisioning, shadow, OTA, and jobs.    I did the work that Espressif should have done but didn't.   They pulled demos from AWS and their OTA sample is a completely different animal and is not cohesive to the basic demos.   I managed to merge it all together.  It took months of work to get everything to go right for me.

I just don't give that stuff away.... sorry.

I built my files one step at a time pulling from the demos and samples.   You just shouldn't short cut that investigation because you will need some understanding of what is going there.

You'll get it.

bakermat, 2022-06-05:
Fair enough. Are you using the latest `esp-iot-aws` release by the way? I can see similar code to yours in the previous branch as @twlawrenceko, not in the current one.

The current `ota_mqtt` example doesn't even compile in my case so basing it off that seems less than efficient.

SolidStateLEDLighting, 2022-06-05:
I pulled my code from what was considered the beta about Nov/Dec of last year.   That beta was supposed to be released (according to an Espressif press release), but that was just a big fat lie.

I do not know what they have released from that point.   I stopped working on the project late last year.

If you can get subscriptions and publishing to work, you have all you need to get everything going.   Everything is really based on just those simple elements, once you have all the policies, roles, templates, etc set up correctly over at AWS.

Just incrementally add all that you need step by step and work your way through it.

Job is a bit more involved, but again it boils down to subscriptions/unsubscriptions and publishing.

twlawrenceko, 2022-06-06:
@bakermat , I am using the ota_mqtt example as a base as that includes the most library dependencies that are required for fleet provisioning. 

@SolidStateLEDLighting , can you provide some information on if you have used the `posix` (not `posix_compat`) library for your fleet provisioning? 

`mbedtls_pkcs11_posix.c` seems to be a self written package from AWS. Did you use this library?

I am using the `fleet_provisioning_with_csr` example from AWS embedded C repo but cannot resolve figuring out linking the correct library for `mbedtls_pkcs11_posix.h`

```
../main/mqtt_operations.c:51:10: fatal error: mbedtls_pkcs11_posix.h: No such file or directory
 #include "mbedtls_pkcs11_posix.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
```

SolidStateLEDLighting, 2022-06-06:
I have not embraced posix.   I don't pay any attention to this kind of compatibility.    I believe the IDF provides some (or much) of those routines in a simpler form.    I didn't care for -- what looked like more complex methods within posix libraries.

mbedtls_pkcs11_posix is not within my project.

Notice:   I do all the normal MQTT security things -- but when using OTA, I do not use encryption.    I download the binary in an unencrypted form (I tricked AWS).    I don't go through the last 5 to 7 unencryption steps when unpacking that binary.    I just installed it as download --- so, I'm not sure just how much I rely on the pksc11 libraries.

lawrenceko-intecular, 2022-06-08:
I have attempted to follow the example as @twlawrenceko has mentioned, where when processing `MQTT_ProcessLoop()` will cause LoadProhibited error to show up. The exact [line](https://github.com/lawrenceko-intecular/esp-aws-iot-for-fleet-provisioning/blob/b4e7158efaa831131e1b32208d450fb828c3714f/examples/fleet_prov_simple/main/shadow_demo_main.c#L1019)

I have read some issues mentioning the buffer size, currently is set to 2048 and still crashing. However, when commented out the publishing topic, the crash occurs less often but still happens overtime.

Below is the log:
```
Info: Establishing MQTT session with claim certificate...
I (5603) coreMQTT: Establishing a TLS session to a3s528xxxxxx4b-ats.iot.us-east-1.amazonaws.com:8883.
I (7763) coreMQTT: Packet received. ReceivedBytes=2.
I (7763) coreMQTT: CONNACK session present bit set.
I (7763) coreMQTT: Connection accepted.
I (7763) coreMQTT: Received MQTT CONNACK successfully from broker.
I (7773) coreMQTT: MQTT connection established with the broker.
I (7783) coreMQTT: MQTT connection successfully established with broker.
I (7793) coreMQTT: An MQTT session with broker is re-established. Resending unacked publishes.
Info: Established connection with claim credentials.
I (7803) coreMQTT: SUBSCRIBE topic $aws/certificates/create/json/accepted to broker.
I (7963) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
I (10963) coreMQTT: SUBSCRIBE topic $aws/certificates/create/json/rejected to broker.
I (11143) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
I (13403) coreMQTT: SUBSCRIBE topic $aws/provisioning-templates/IVO1-FleetProvisioning/provision/json/accepted to broker.
I (13533) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
I (16533) coreMQTT: SUBSCRIBE topic $aws/provisioning-templates/IVO1-FleetProvisioning/provision/json/rejected to broker.
I (16673) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
I (19673) coreMQTT: Published payload: 
I (19673) coreMQTT: PUBLISH sent for topic $aws/certificates/create/json to broker with packet ID 5.
I (19843) coreMQTT: Packet received. ReceivedBytes=2.
I (19843) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (19853) coreMQTT: State record updated. New state=MQTTPublishDone.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c619  PS      : 0x00060f30  A0      : 0x800da214  A1      : 0x3ffbb350  
A2      : 0x00000000  A3      : 0x3f4065e8  A4      : 0x00000022  A5      : 0x0000ff00  
A6      : 0x0000000f  A7      : 0xff000000  A8      : 0x00000002  A9      : 0x3ffbb000  
A10     : 0x3ffbb020  A11     : 0x3f405952  A12     : 0x3ffbb284  A13     : 0x3ffaf02c  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000002  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace:0x4000c616:0x3ffbb3500x400da211:0x3ffbb370 0x400da330:0x3ffbb390 0x400da434:0x3ffbb3c0 0x400d70b6:0x3ffbb3e0 0x400d8b41:0x3ffbb410 0x400d8bd6:0x3ffbb450 0x400d8cd7:0x3ffbb480 0x400d8d15:0x3ffbb4b0 0x400d775d:0x3ffbb4d0 0x400d7202:0x3ffbb4f0 0x400d72a4:0x3ffbb510 0x400d70a5:0x3ffbb530 0x40156b1d:0x3ffbb550 0x4008b525:0x3ffbb570 
0x400da211: consumeIfMatch at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/source/fleet_provisioning.c:587

0x400da330: parseCreateCertificateFromCsrTopic at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/source/fleet_provisioning.c:458

0x400da434: FleetProvisioning_MatchTopic at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/source/fleet_provisioning.c:744

0x400d70b6: provisioningPublishCallback at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/examples/fleet_prov_simple/build/../main/shadow_demo_main.c:870

0x400d8b41: handlePublishAcks at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/coreMQTT/coreMQTT/source/core_mqtt.c:1195

0x400d8bd6: handleIncomingAck at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/coreMQTT/coreMQTT/source/core_mqtt.c:1241

0x400d8cd7: receiveSingleIteration at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/coreMQTT/coreMQTT/source/core_mqtt.c:1342

0x400d8d15: MQTT_ProcessLoop at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/libraries/coreMQTT/coreMQTT/source/core_mqtt.c:2185

0x400d775d: ProcessLoop at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/examples/fleet_prov_simple/build/../main/shadow_demo_helpers.c:941

0x400d7202: waitForResponse at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/examples/fleet_prov_simple/build/../main/shadow_demo_main.c:776

0x400d72a4: aws_iot_demo_main at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/examples/fleet_prov_simple/build/../main/shadow_demo_main.c:1019

0x400d70a5: app_main at /home/espuser/esp-public-git/esp-aws-iot-for-fleet-provisioning/examples/fleet_prov_simple/build/../main/app_main.c:58 (discriminator 2)

0x40156b1d: main_task at /home/espuser/esp/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x4008b525: vPortTaskWrapper at /home/espuser/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

SolidStateLEDLighting, 2022-06-08:
If you count all the characters from a provisioning response, it will be around 3,500.   Set your buffer to at least 4k (4096).   Give that a try.   Buffer over-flows don't crash the application, you just don't receive the data.  This LoadProhibited error is from something else -- like a bad pointer.

Remember to always circle back around to MQTT_ProcessLoop() for the results of each action (construct a large state driven super loop).   You will receive a response in the loop for pretty much everything you do.

lawrenceko-intecular, 2022-06-08:
We have also noticed that is is around 3500, but the idf.py menuconfig limits to 2048, so we should manually change it to 4096?

One thing we have is this receiving messages after each subscribe, is this correct behavior?
```
I (14213) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
```

SolidStateLEDLighting, 2022-06-08:
HEY -- One more thing --- increase your stack size in that IOT task handler to about 16k (16,384) and examine if you like the HighWaterMark to analyze your memory usage as needed.

If you run out of memory in the task handler, you will crash.

SolidStateLEDLighting, 2022-06-08:
If my memory serves me well today, that 2048 limit is just set by another file in the system.   Do a search in files for "2048" and you'll find it -- change it.

SolidStateLEDLighting, 2022-06-08:
In addition to a successful result in your subscriptions/publishing -- you should always receive back a message and that message will only be received in the MQTT_ProcessLoop().    Always return to that loop between actions and handle all incoming message or error results.

SolidStateLEDLighting, 2022-06-08:
HEY -- this error message says:      Warn: Unexpected publish message received. Topic: .

You sent a publish message that was unexpected....

And the topic was "."

So, you may not have been subscribed to anything?... and you didn't send a full string message.

lawrenceko-intecular, 2022-06-08:
> HEY -- this error message says: Warn: Unexpected publish message received. Topic: . You sent a publish message that was unexpected.... And the topic was "." So, you may not have been subscribed to anything?... and you didn't send a full string message.
> […](#)

The log shows 
```
I (14325) coreMQTT: Packet received. ReceivedBytes=3.
Warn: Unexpected publish message received. Topic: .
```

So I thought it was receiving messages from AWS IoT, not me publishing other topics to AWS IoT.

SolidStateLEDLighting, 2022-06-08:
Oh... yea... your hardware received 3 bytes identified as a response to your publish request and it is so small that it doesn't specify the Topic responded to.

lawrenceko-intecular, 2022-06-08:
The LoadProhibited error occurs when I am trying to print out the JSON payload. I have increased the main task heap size to 8192, is there any other portion that needs to be taken care of giving more space?

SolidStateLEDLighting, 2022-06-08:
I have my IOT task heap set to 16k.

I can't quite recall why I set it up that high, except to say that I experimented to come up that number.

Larger is better to get the thing running and then determine how much you can reduce later.

lawrenceko-intecular, 2022-06-09:
@SolidStateLEDLighting , how did you import the certificates and private keys into PKCS11Session?

```
/* Insert the claim credentials into the PKCS #11 module */
            pkcs11Status = loadClaimCredentials( p11Session,
                                                CLAIM_CERT_PATH,
                                                "Claim Cert",
                                                CLAIM_PRIVATE_KEY_PATH,
                                                "Claim Key" );
```
This is what AWS embedded C has provided, however, within ESP, the path would not be valid as it cannot find where the certificate is located.

SolidStateLEDLighting, 2022-06-09:
I do not recognize the loadClaimCredentials command that you use here.

What is the loadClaimCredentials doing for you?

In my work the provisioning credentials arrive in a text file and are immediately usable.

Why do the credentials need to be loaded into PKCS11?

[7 more replies]