espressif/esp-aws-iot #91: Memory leak in TLS_FreeRTOS_Connect/Disconnect

TakuyaAtMono, 2022-02-09:
I am using release/beta branch port/network_transport to make the TLS connection.
When I repeat "TLS_FreeRTOS_Connect" and "TLS_FreeRTOS_Disconnect" in this source code, the available memory space decreases.

I checked the source code and found that in "tls_freertos.c", line 67 It seems that the area allocated by esp_transport_list_init has not been released.
If it is managed by a list, shouldn't we execute esp_transport_list_destroy instead of esp_transport_destroy in line 135 when do TLS_FreeRTOS_Disconnect?

Please let me know what is the correct way to handle this.

I have checked the available memory size with heap_caps_get_free_size.
Also, the capacity of heap_caps_get_largest_free_block is reduced.

shahpiyushv, 2022-02-10:
@TakuyaAtMono , thanks for reporting this. The issue has been fixed.

closed 2022-02-10 by shahpiyushv

TakuyaAtMono, 2022-02-14:
Thank you for your quick response.
I have confirmed that it has been fixed.
However, the same process is done when the connection fails, so I guess it should be fixed that too. (tls_freertos.c line 110)