espressif/esp-aws-iot #93: Porting of AWS SDK to use S3 bucket features (IDFGH-6734)

MaazSk, 2022-02-17:


MaazSk, 2022-02-23:
Hi @mahavirj,

I just wanted to know if there are any updates on it?

MaazSk, 2022-03-05:
Hi @dhavalgujar,

Thank you for your valuable response. I will check these examples and update you soon as I am preoccupied right now and if any help is needed I will let you know for sure.

Thank you for sharing the details

MaazSk, 2022-03-16:
Hi @dhavalgujar, 
Can you please guide me on how to build the above-mentioned examples as I am unable to build them?

MaazSk, 2022-03-17:
Hi @dhavalgujar,

I was going through the examples, but those examples seem to be for pre-signed URLs of AWS. Is there an example where we can upload files without pre-signed URLs?

MaazSk, 2022-04-14:
Hi @mahavirj, is there any update on this?

Thanks and Regards, 
Maaz Shaikh

dizcza, 2022-12-18:
> Hi, @MaazSk
> 
> We have made the port for the coreHTTP library and an example to go with it available [here](https://github.com/espressif/esp-aws-iot/tree/release/beta/examples/http/http_mutual_auth). The example demo that we have made available has been taken from, and is a lightly changed version of the one available [in the aws-iot-device-sdk-embedded-C repo](https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/59ecfed9fc03076984506d6218256718ab1d4ed7/demos/http/http_demo_mutual_auth).
> 
> The same external repo includes examples for S3 [upload](https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/59ecfed9fc03076984506d6218256718ab1d4ed7/demos/http/http_demo_s3_upload) and [download](https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/59ecfed9fc03076984506d6218256718ab1d4ed7/demos/http/http_demo_s3_download).
> 
> Since the library used for the demos is the same and has been ported, and there are no other dependencies for S3 upload specifically, it should be possible to make similar changes to the example, and get the S3 upload demo to work on esp32. You can view a diff of the changes made to the original example [here](https://www.diffchecker.com/p6y94BWr).
> 
> Please feel free to reply here in case you run into any issues with getting the S3 upload demo to work.
> 
> HTH.

Any reason why the code in the `release/beta` has not been merged into master? It's a bad practice to stick to an unstable version and I see the branches have already diverged significantly.