espressif/esp-aws-iot #94: esp ota mqtt

git4gaurav, 2022-02-17:
hi, i'm reffering to:
https://github.com/espressif/esp-aws-iot/tree/release/beta/examples/ota/ota_mqtt
By using esp-idf eclipse v4.20. the below error is obtained is attached
[log1.txt](https://github.com/espressif/esp-aws-iot/files/8088929/log1.txt)
please suggest how to resolve this. currently files downloaded and build. no aws/device side configurations made.

EmbeddedDBL, 2022-02-18:
I recommend that you use a version of esp-idf equal to or greater than 4.3 .
You could start by checking if you have FreeRTOS static location enabled.
idf.py menuconfig -> Component config -> FreeRTOS -> Enable FreeRTOS static alloction API

git4gaurav, 2022-02-18:
Hi... Im having ESP IDF v4.3
I can see 
idf.py menuconfig -> Component config -> FreeRTOS -> Enable FreeRTOS static alloction API
Im attaching screenshot of above.
[Config.docx](https://github.com/espressif/esp-aws-iot/files/8094640/Config.docx)

Pl. suggest what can be done

git4gaurav, 2022-02-23:
Below are the errors obtained on build:
[972/1009] ccache C:\Users\gau\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../../../../aws-iot-device-sdk-embedded-C/demos/ota/common/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/portable/os -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/backoffAlgorithm/source/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/portable -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/dependency/coreJSON/source/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/aws/ota-for-aws-iot-embedded-sdk/source/dependency/3rdparty/tinycbor/src -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/interface -I../../../../port/logging -I../../../../port/network_transport -I../../../../port/posix-compat -I../../../../aws-iot-device-sdk-embedded-C/platform/include -I../../../../port/crypto -I../../../../port/ota -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/dependency/3rdparty/pkcs11 -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/include -I../../../../aws-iot-device-sdk-embedded-C/libraries/standard/corePKCS11/source/dependency/3rdparty/mbedtls_utils -I../../../../../../newlib/platform_include -I../../../../../../freertos/include -I../../../../../../freertos/port/xtensa/include -I../../../../../../esp_hw_support/include -I../../../../../../esp_hw_support/port/esp32/. -I../../../../../../heap/include -I../../../../../../log/include -I../../../../../../lwip/include/apps -I../../../../../../lwip/include/apps/sntp -I../../../../../../lwip/lwip/src/include -I../../../../../../lwip/port/esp32/include -I../../../../../../lwip/port/esp32/include/arch -I../../../../../../soc/include -I../../../../../../soc/esp32/. -I../../../../../../soc/esp32/include -I../../../../../../hal/esp32/include -I../../../../../../hal/include -I../../../../../../esp_rom/include -I../../../../../../esp_rom/esp32 -I../../../../../../esp_rom/include/esp32 -I../../../../../../esp_common/include -I../../../../../../esp_system/include -I../../../../../../esp32/include -I../../../../../../driver/include -I../../../../../../driver/esp32/include -I../../../../../../esp_ringbuf/include -I../../../../../../efuse/include -I../../../../../../efuse/esp32/include -I../../../../../../xtensa/include -I../../../../../../xtensa/esp32/include -I../../../../../../espcoredump/include -I../../../../../../esp_timer/include -I../../../../../../esp_ipc/include -I../../../../../../esp_pm/include -I../../../../../../vfs/include -I../../../../../../esp_wifi/include -I../../../../../../esp_wifi/esp32/include -I../../../../../../esp_event/include -I../../../../../../esp_netif/include -I../../../../../../esp_eth/include -I../../../../../../tcpip_adapter/include -I../../../../../../app_trace/include -I../../../../../../mbedtls/port/include -I../../../../../../mbedtls/mbedtls/include -I../../../../../../mbedtls/esp_crt_bundle/include -I../../../../../../app_update/include -I../../../../../../spi_flash/include -I../../../../../../bootloader_support/include -I../../../../../../nvs_flash/include -I../../../../../../pthread/include -I../../../../../../esp_gdbstub/include -I../../../../../../esp_gdbstub/xtensa -I../../../../../../esp_gdbstub/esp32 -I../../../../../../wpa_supplicant/include -I../../../../../../wpa_supplicant/port/include -I../../../../../../wpa_supplicant/include/esp_supplicant -I../../../../../../perfmon/include -I../../../../../../asio/asio/asio/include -I../../../../../../asio/port/include -I../../../../../../cbor/port/include -I../../../../../../unity/include -I../../../../../../unity/unity/src -I../../../../../../cmock/CMock/src -I../../../../../../coap/port/include -I../../../../../../coap/port/include/coap -I../../../../../../coap/libcoap/include -I../../../../../../coap/libcoap/include/coap2 -I../../../../../../console -I../../../../../../nghttp/port/include -I../../../../../../nghttp/nghttp2/lib/includes -I../../../../../../esp-tls -I../../../../../../esp-tls/esp-tls-crypto -I../../../../../../esp_adc_cal/include -I../../../../../../esp_hid/include -I../../../../../../tcp_transport/include -I../../../../../../esp_http_client/include -I../../../../../../esp_http_server/include -I../../../../../../esp_https_ota/include -I../../../../../../protobuf-c/protobuf-c -I../../../../../../protocomm/include/common -I../../../../../../protocomm/include/security -I../../../../../../protocomm/include/transports -I../../../../../../mdns/include -I../../../../../../esp_local_ctrl/include -I../../../../../../sdmmc/include -I../../../../../../esp_serial_slave_link/include -I../../../../../../esp_websocket_client/include -I../../../../../../expat/expat/expat/lib -I../../../../../../expat/port/include -I../../../../../../wear_levelling/include -I../../../../../../fatfs/diskio -I../../../../../../fatfs/vfs -I../../../../../../fatfs/src -I [... cut ...] es/__idf_main.dir/__/__/__/__/aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/core_mqtt_state.c.obj   -c C:/esp-idf/components/ota_beta_esp32/esp-aws-iot/aws-iot-device-sdk-embedded-C/libraries/standard/coreMQTT/source/core_mqtt_state.c
ninja: build stopped: subcommand failed.
Build complete (0 errors, 0 warnings): C:\esp-idf\components\ota_beta_esp32\esp-aws-iot\examples\ota\ota_mqtt\build
Total time taken to build the project: 178,771 ms

please let me know how to resolve this

closed 2022-03-23 by git4gaurav