espressif/esp-aws-iot #96: example download

git4gaurav, 2022-02-21:
Hi,
the ota mqtt demo for aws meant for esp32 was downloaded from :
https://github.com/espressif/esp-aws-iot/tree/release/beta/examples/ota/ota_mqtt

Steps use to download project files were: 
1. download .zip file from:
https://github.com/espressif/esp-aws-iot/tree/release/beta
2. Download libraries using:
git clone --recurse-submodules https://github.com/aws/aws-iot-device-sdk-embedded-C.git

following are the issues faced:
a. there is no clone link for ota mqtt demo
b. when .zip is downloaded, library/ sdk files are missing

Please check attached file having error log on build. Pl. suggest if file download steps are ok and help in resolving build errors.
Problem window image is shown below:
[image]

shahpiyushv, 2022-02-21:
@git4gaurav , if you download the repository as a zip, the git commands, including `git submodule update --init --recursive` would not work. I suppose you may have got some error like `fatal: not a git repository (or any of the parent directories): .git`. Please download using `git clone --recursive -b release/beta https://github.com/espressif/esp-aws-iot`

git4gaurav, 2022-02-22:
Hi, below error image is shown... 
[image]
pl. let me know, what should be done to resolve above

dhavalgujar, 2022-03-01:
> and when device changed in edit to esp32-c3....then below is the error: Description Resource Path Location Type fatal error: soc/dport_reg.h: No such file or directory esp32-c3-ota line 37, external location: C:\esp-idf\components\ota-esp32c3\esp-aws-iot\port\ota\aws_esp_ota_ops.c C/C++ Problem
> 
> What are necessary configurations required in config aprt from WiFi
> 
> Please guide on this.

Can you please try the patch mentioned [here](https://github.com/espressif/esp-aws-iot/issues/99#issuecomment-1049100846) and confirm if it fixes it for you?
This will be fixed in the port shortly.

git4gaurav, 2022-03-03:
Thanks shahpiyushv and dhavalgujar. I was able to run demo.

closed 2022-03-03 by git4gaurav