espressif/esp-aws-iot #99: Request OTA example compatibility for ESP32-C3

iceananas, 2022-02-23:
## Expected Behavior

The OTA MQTT example is expected to build for esp32-c3 target.

## Current Behavior

Building the example fails because `/port/ota/aws_esp_ota_ops.c` includes a header which the esp32-c3 does not have.

## Steps to Reproduce

1. Clone repository and go to `examples/ota/ota_mqtt`
2. Switch target using `idf.py set-target esp32c3`
3. Try to build the taget `idf.py build`
4. The build fails with an error 
```
..../esp-aws-iot/port/ota/aws_esp_ota_ops.c:37:10: fatal error: soc/dport_reg.h: No such file or directory
 #include "soc/dport_reg.h"
```

## Context (Environment)
ESP-IDF version 4.3.2

dhavalgujar, 2022-02-23:
Hi @iceananas, thank you for reporting this.
Please try the attached patch.
[c3-compatibility.txt](https://github.com/espressif/esp-aws-iot/files/8126989/c3-compatibility.txt)
It will be fixed in the port shortly.

iceananas, 2022-02-23:
Well that was fast :) I can confirm that the example builds with the patch. Thanks!