espressif/esp-aws-iot #101: soc/dport_reg.h missing esp-idf 4.4 for esp32-c3, CDT Code Builder Error

YogeshG30, 2022-03-01:
Dear All,

We are working on BLE Wi-Fi Gateway using ESP32-C3 (ESp32-C3-DevKitM-1).

While working on esp-idf 4.4 and Esp-aws-iot, we came across following 2 observation

**Observation 1: soc/dport_reg.h missing esp-idf 4.4 for esp32-c3**
     We have installed esp-idf from "Epressif-IDE 2.4.0 with ESP-IDF 4.4 Windows 10, size 1GB" option of esp-idf windows installer link.
 
     For ota_mqtt & tls_mutual_authentication reference code, functionality we are using latest Esp-aws-iot from below link
https://github.com/espressif/esp-aws-iot/tree/release/beta

     We found tls_mutual_authentication compiling and working correctly for ESP32 and ESP32-C3

     We found ota_mqtt compiling correctly for ESP32. But it is giving file missing error for ESP32-C3 as below.
C:/Project/Goodsman/Source/esp-aws-iot/port/ota/aws_esp_ota_ops.c:37:10: fatal error: soc/dport_reg.h: No such file or directory
#include "soc/dport_reg.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Build complete (1 errors, 0 warnings): C:\Project\Goodsman\Source\esp-aws-iot\examples\ota\ota_mqtt\build
Our queries:
   1.	How we can overcome this file missing error for ESP32-C3
   
**Observation 2 : CDT Core Builder error:**
             
            Me and my 2 colleagues installed ESP-IDF 4.4 as mentioned in above query.
 
            While compiling ota_mqtt & tls_mutual_authentication reference code on ESP32, I am not getting any error.
             
            One of my colleague facing following error while compiling the code as attached. But he is able to compile the code using “ESP-IDF 4.4 PowerShell”.
 
            Our queries:
           1.	What are the possible error for CDT Code Builder Error?
           3.	How to overcome this error?
 
[image]

Thanks....

dhavalgujar, 2022-03-01:
Can you please try the patch mentioned [here](https://github.com/espressif/esp-aws-iot/issues/99#issuecomment-1049100846) and confirm if it fixes it for you?
This will be fixed in the port shortly.

YogeshG30, 2022-03-02:
Yes the patch worked for us. The code is now compiling.
Thanks for the support.