espressif/esp-aws-iot #104: Query about Build System Command idf.py size-components

YogeshG30, 2022-03-03:
Dear all,

We are working on Gateway project using ESP32-C3.

We are working on reference code for various Gateway components like AWS connectivity, Device connectivity over BLE, OTA, Web server.

We are analyzing memory analysis for each component using **idf.py size-components** command to estimate overall memory requirement for all components together.

We have few queries for output of idf.py size-components.

Find below the output for OTA_mqtt reference code 

[image]

query 1 - We understood libmain is Main application. Is our understanding correct?

query 2 - We observed the RAM and Flash size for libmain of OTA_mqtt reference code is higher than other reference code like BLE, Webserver. We understood it is because the size includes the CSDK libraries like IOT_mqtt, IOT_ota, etc. Is our understanding correct?

We found used flash size for OTA_mqtt reference code as below
[image]
query 3: .text + .rodata comes 850KB. But **Used Flash Size** showing 1571 KB. How Used Flash Size derived?

query 4 : what is the content of libdriver ?

Regards...

Yogesh

ameyinamdar, 2022-03-09:
Yogesh, you are not considering IRAM text, DRAM data and padding that gets added into the image. If you see "idf.py size" output, you'll see something like:

Total sizes:
Used stat D/IRAM:   99160 bytes ( 228520 remain, 30.3% used)
      .data size:   19084 bytes
      .bss  size:   20232 bytes
      .text size:   59844 bytes
Used Flash size :  805986 bytes
      .text     :  662416 bytes
      .rodata   :  143314 bytes
Total image size:  884914 bytes (.bin may be padded larger)

(note: this is for a different example)

YogeshG30, 2022-03-09:
Hi Amey,

Thanks for the clarification.
In your example, .text = 662416 + .rodata = 143314 comes 8,05,730 which is matching with Used Flash size : 805986 bytes

[image]
In our case as shown above, Flash total (Highlighted orange) we understood addition of DRAM.data + .rtc.data + IRAM0.text + Flash.text + & .rodata + & .appdesc & .rodata_noload
So  flash size should be 920318

.text +   .rodata (Highlighted Yellow) comes 850668. If we add DRAM.data +   rtc.data+IRAM0.text then it comes 923135 which is close to 920318 expected.

But total Used Flash size is 1571820 (Highlighted Blue). Why it is not coming around 920318?

Regards...

Yogesh