espressif/esp-aws-iot #105: Request OTA example compatibility for *ESP32-S3*

tedevel, 2022-03-10:
Just as issue "Request OTA example compatibility for ESP32-C3 (CA-196) #99" I'd like to request a similar patch to the c3-compatibility.txt but for the ESP32 S3.

I've confirmed that this patch words for the ESP32-C3, was hopeful it would also get the S3 to build, but it did not.

A patch for this would be great :) OTA compatibility central to production development with the S3!

andrew-elder, 2022-04-15:
I'm looking for the same thing!

andrew-elder, 2022-04-15:
Here is patch that I worked out that allows S3 to compile
[esp32s3-compatility.txt](https://github.com/espressif/esp-aws-iot/files/8496995/esp32s3-compatility.txt)

I think it is correct, but I can't guarantee it.
.