espressif/esp-aws-iot #106: aws_iot_mqtt_subscribe() fails with NETWORK_SSL_READ_ERROR

sduddell, 2022-03-11:
Hi,

I'm trying to subscribe to an MQTT topic using aws_iot_mqtt_subscribe(), but it fails.

As shown in below code, trying to subscribe to topic `"M5TimerCameraF1/pub" ` :

```
    string c2a_topic = "M5TimerCameraF1/pub";
    const int topic_len = c2a_topic.length();

    ESP_LOGI(AWS_IOT_TAG, "Subscribing...");
    result = aws_iot_mqtt_subscribe(&aws_client, c2a_topic.c_str(), topic_len, QOS0, iot_subscribe_callback_handler, NULL);
    if(SUCCESS != result) {
        ESP_LOGE(AWS_IOT_TAG, "Error subscribing : %d ", result);
        abort();
    } 
```
It fails as shown below:
```

␛[0;32mI (22295) AwsIoT: Subscribing...␛[0m
␛[0;31mE (22505) aws_iot: Failed
  ! mbedtls_ssl_read returned -0x7880

␛[0m
␛[0;31mE (22505) AwsIoT: Error subscribing : -12 ␛[0m
```

It connected properly to AWS IOT before subscribing though.
Are there any pre-requisites to be done before subscription?
Any suggestions/solutions would help.

Thanks!

shahpiyushv, 2022-03-11:
@sduddell , does your AWS IoT Policy allow the client to subscribe to this topic?

sduddell, 2022-03-11:
Hi Piyush,

Thanks for quick reply.
Yes, AWS IoT policy is allowing, still this error is seen.
[image]

MaazSk, 2022-03-29:
Hi @sduddell, did you found the solution as I am facing the same issue.

sduddell, 2022-03-29:
In my case, certificate in the device was different from policy where I'm allowing it to connect.
You can also check the cloudwatch logs to understand the error from AWS IoT perspective.

closed 2022-03-29 by sduddell