espressif/esp-aws-iot #108: NETWORK_RECONNECT_TIMED_OUT_ERROR after no internet

MaazSk, 2022-03-22:
I am using esp-idf v 4.3. I am facing the following issue 
When there is no internet for a certain duration of time.  The mqtt connection gets disconnected and when the internet is back the device is not connected to the MQTT and even if I try ```aws_iot_mqtt_attempt_reconnect()```. It does not connect back and after a certain time I am getting this error ``` NETWORK_RECONNECT_TIMED_OUT_ERROR```. How can I reconnect to MQTT?

Also, I am suspending the aws task when there is no internet and when the internet connection is back I am resuming the aws task.

Do I need to destroy aws connection and then reconnect?

Or is there a workaround solution for this problem

Thank you!

devUsama, 2022-04-12:
One approach which I am using is that i have disabled the mqtt_autoreconnect functionality and whenever client needs to connect with broker after disconnection i manually call aws_iot_mqtt_attempt_reconnect() until it gets connected again. 

But in some scenarios like in my case sometimes aws_iot_mqtt_yield() returns NETWORK_SSL_READ_TIMEOUT_ERROR, in that case i have to disconnect and destroy the aws connection and then reconnect it again to make it work.

That is the APIs flow for reference:
 aws_iot_mqtt_disconnect(&mqttClient); 
 aws_iot_mqtt_free(&mqttClient);

  // initiate connection again with client
  rc = aws_iot_mqtt_connect(mqttClient, &cp);

MaazSk, 2022-04-14:
@devUsama Okay, the same approach is been used by me now.