espressif/esp-aws-iot #109: AWS IoT last will does not seem to be working

DarshanDodal, 2022-03-24:
I am using the shadow example to connect with my aws iot thing and everything is working fine up to this.  However, when I try using the lwt configuration (I have updated the "mqttClient" struct from example with the last will configurations), it does not send any message on the specified topic. 

This is how I have set the mqttClient Struct from the example. 

AWS_IoT_Client mqttClient;

 char lwtopic[MAX_SHADOW_TOPIC_LENGTH_BYTES];
 snprintf(lwtopic, MAX_SHADOW_TOPIC_LENGTH_BYTES,"things/%s/shadow/update", THING_NAME);
 char sid[4];
  snprintf(mqttClient.clientData.options.will.struct_id, 4,"69");
  // mqttClient.clientData.options.will.struct_id=sid;
  mqttClient.clientData.options.will.pTopicName=lwtopic;         
  mqttClient.clientData.options.will.topicNameLen=strlen(lwtopic);                       
  mqttClient.clientData.options.will.pMessage="{\"state\":{\"desired\":{\"s\":false},\"reported\":{\"s\":false}}}";                                   
  mqttClient.clientData.options.will.msgLen=strlen("{\"state\":{\"desired\":{\"s\":false},\"reported\":{\"s\":false}}}");         
  mqttClient.clientData.options.will.isRetained=false;                     
  mqttClient.clientData.options.will.qos=1;

shahpiyushv, 2022-03-25:
@DarshanDodal , have you checked if your Thing's AWS IoT policy allows publishing to this topic?

DarshanDodal, 2022-03-25:
Yes. I have not set any restrictions for specific topics in the policy, all topics are simply allowed.

shahpiyushv, 2022-03-25:
@DarshanDodal , not sure how it matters, but the  AWS IoT docs [here](https://github.com/espressif/aws-iot-device-sdk-embedded-C/blob/d1417908038fed0cc68b753a43fca3ea82f97917/include/aws_iot_mqtt_client.h#L114) say that the struct_id must be MQTW. Have you tried that?

DarshanDodal, 2022-03-25:
Oh, that is where I did a mistake I guess.
Let me try that!

DarshanDodal, 2022-03-26:
@shahpiyushv what if I use lifecycle events of aws iot core, instead of specifying the LWT at device end?. Will that make any difference?

shahpiyushv, 2022-03-28:
@DarshanDodal , lifecycle events have pre defined special topics and payloads. If they suit your use case, you may use them. However, if you want to use custom topics and payloads, LWT would be better.

shahpiyushv, 2022-03-30:
Closing this issue since it has been answered.

closed 2022-03-30 by shahpiyushv