espressif/esp-aws-iot #115: Queries on OTA, mqtt_demo, BLE central, Webserver

YogeshG30, 2022-04-11:
Hi All,

We are working on BLE – Wi-Fi Gateway using ESP32-C3 with features like Webserver, BLE Central, AWS data upload & fetch, OTA from AWS.
We are using latest ESP IDF 2.4.1
We referred following code from AWS IoT Device SDK for Embedded C 202103.00 
•	mqtt_demo_mutual_auth
•	ota_demo_core_mqtt
•	BLE Bludroid GATT Central
We integrated all above code together. We have few observations and queries
1)	For mqtt_demo_mutual_auth
Observation: In case gateway connected to router over Wi-Fi with Internet connection not present to router, on every failure attempt of connectToServerWithBackoffRetries, the connectToServerWithBackoffRetries function leak around 1k of RAM
Query: How leak memory of around 1K for failure attempt of connectToServerWithBackoffRetries can be avoided?

2)	For integrated code
      Observation: The remaining heap memory for integrated code after all task invocation is around 16K. 

      Query: Is it ok to have free heap memory of 16K

3)	For integrated code
Observation:  The remaining heap memory for integrated code after all task invocation is around 16K. During Wifi/Internet disconnect and reconnect, connectToServerWithBackoffRetries did not connect with error esp-tls: could not get host name for: XXX getaddrinfo() return 202

               Query: We guess the error due to shortage of memory. Is our understanding correct? How can we overcome this error?

4)	For integrated code
      Observation: With integrated code without OTA, free heap memory is 134 k. With OTA integration it comes down to 16 K.

      Query: What are different ways, we can reduce memory consumption for OTA?  

5)	For mqtt_demo_mutual_auth, ota_demo_core_mqtt integration
      Observation: 
•	both mqtt_demo_mutual_auth, ota_demo_core_mqtt sample code are using separate network_context and mqtt_context. 
•	mqtt_demo_mutual_auth disconnect network_context and mqtt_context after data upload
•	ota_demo_core_mqtt maintain same throughout operation
•	For integration, we used network_context and mqtt_context initiated by ota_demo_core_mqtt in mqtt_demo_mutual_auth. 
o	We found same working correctly. 
o	We commented network_context and mqtt_context in mqtt_demo_mutual_auth
o	network_context and mqtt_context connection and disconnection code in mqtt_demo_mutual_auth is commented

      Query: The way we handle network_context and mqtt_context for integration is correct?

6)	For QOS1 implementation for Publish
      Observation: Instead of using pSessionPresent, handlePublishResend, we are planning to use simple PUB_ACK check after publish and retry for 3 attempts.

      Query: Is it ok to have logic for QOS1 as we mentioned above

Thanks.....

ameyinamdar, 2022-04-13:
@YogeshG30  we recently moved LTS support to master branch with many changes. Can you please try with the same? We'll also make the integrated example available shortly.

YogeshG30, 2022-04-14:
Hi Amey,

Thanks for the reply.

Currently we are using CSDK beta release as below for which we post queries above.
https://github.com/espressif/esp-aws-iot/tree/release/beta

As you mentioned we tried master branch OTA code from below link
https://github.com/espressif/esp-aws-iot/tree/master/examples/ota/ota_mqtt

With beta release we are having following observation
1) When device Wi-Fi connection disconnected and restored, no memory leak observed
2) When device Wi-Fi not disconnected and only Internet connection disconnected and restored, memory leak observed around 3 KB

With master branch code we observed memory leak for both Wi-Fi and Internet disconnection and reconnection as below.
WiFi connected and internet present. heap = 103 kB
disconnected Wi-Fi, heap = 140 kB seen once
second time TLS connect = 129 kB
Wi-Fi restored, heap = 83 kB

WiFi connected and internet present, heap = 105 kB
disconnected internet and TLS attempt after keepalive timeout, heap = 137 kB seen once
second time tls attempt= 126 kB
third = 115 kB
fourth = 104 kB
internet restored, heap = 61 kB

So it looks like Beta version is quite stable compared to master version.

On beta version we observed one patch for memory leak implemented 15 days ago. We implemented same. With this all memory leak observation got resolved except one which we mentioned above for 3KB during internet disconnection and restored.

**Please give your input for way to handle 3KB leak in beta version.

Is there any event available for Wi-Fi present but internet disconnected**? 

**Also give your input on our queries for integrating OTA_mqtt and mqtt_demo_mutual_auth as below**

For integrated code
Observation: With integrated code without OTA, free heap memory is 134 k. With OTA integration it comes down to 16 K. Basis on initial we were expecting 65KB for Control + Data

Query: What are different ways, we can reduce memory consumption for OTA?

For mqtt_demo_mutual_auth, ota_demo_core_mqtt integration
Observation:
• both mqtt_demo_mutual_auth, ota_demo_core_mqtt sample code are using separate network_context and mqtt_context.
• mqtt_demo_mutual_auth disconnect network_context and mqtt_context after data upload
• ota_demo_core_mqtt maintain same throughout operation
• For integration, we used network_context and mqtt_context initiated by ota_demo_core_mqtt in mqtt_demo_mutual_auth.
o We found same working correctly.
o We commented network_context and mqtt_context in mqtt_demo_mutual_auth
o network_context and mqtt_context connection and disconnection code in mqtt_demo_mutual_auth is commented

Query: The way we handle network_context and mqtt_context for integration is correct?

For QOS1 implementation for Publish
Observation: Instead of using pSessionPresent, handlePublishResend, we are planning to use simple PUB_ACK check after publish and retry for 3 attempts.

Query: Is it ok to have logic for QOS1 as we mentioned above

Thanks.....