espressif/esp-aws-iot #119: Bug OTA HTTP

StevenB56, 2022-04-15:
Hi,

I had reported a bug concerning the OTA MQTT which is now solved, I tried the OTA HTTP but failed, see the logs below.
As for the MQTT OTA, the ESP32 reboots following a crash. Thank you for your hard work.

```
I (130857) coreMQTT: Packet received. ReceivedBytes=0.
I (133857) AWS_OTA:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (136857) AWS_OTA:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (139857) AWS_OTA:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (141447) coreMQTT: Packet received. ReceivedBytes=1873.
I (141447) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=M                                                                                                                                                             QTTSuccess.
I (141447) coreMQTT: State record updated. New state=MQTTPublishDone.
I (141457) coreMQTT: Invoking subscription callback of matching topic filter: To                                                                                                                                                             picFilter=$aws/things/+/jobs/#, TopicName=$aws/things/OTA_Device/jobs/notify-nex                                                                                                                                                             t
I (141477) AWS_OTA: Extracted parameter: [key: value]=[execution.jobId: AFR_OTA-                                                                                                                                                             OTA_Job_1404_2222]
I (141497) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr                                                                                                                                                             _ota.protocols: ["HTTP"]]
I (141497) AWS_OTA: Extracted parameter: [key: value]=[filepath: /]
I (141497) AWS_OTA: Extracted parameter: [key: value]=[filesize: 857296]
I (141507) AWS_OTA: Extracted parameter: [key: value]=[fileid: 0]
I (141517) AWS_OTA: Extracted parameter: [key: value]=[certfile: Code Verify Key                                                                                                                                                             ]
I (141527) AWS_OTA: Extracted parameter: [key: value]=[update_data_url: https://                                                                                                                                                             esp32otaawsbucket.s3.eu-west-3.amazonaws.com/SignedImages/15271131-059d-4294-a86                                                                                                                                                             4-dde3184dafe9?X-Amz-Security-Token=IQoJb3JpZ2luX2VjENX%2F%2F%2F%2F%2F%2F%2F%2F%                                                                                                                                                             2F%2FwEaCWV1LXdlc3QtMyJGMEQCIHKr5GWKSDk8cAKXV4%2FAI%2BDV3xr7gBpv3SO1%2FKr1O8FqAi                                                                                                                                                             B8pTyvPK%2F6Ka%2FUwZwKzjpme6OQtfIDta1YwctYfEZ5GSrdAgh%2BEAAaDDYxNTQxMzUxMTQ1NSIM                                                                                                                                                             lrN7t5fOblvjONznKroCKMLkMtXDyQoitWAt2XBnC7O4AS7YGa49a%2BVPcx1WLG37z71r32Q%2FeXrS                                                                                                                                                             C1HwIyD6LsrSXRiddgxgNPzTD0uIcN7b6WtYbbWZmGwaAWm3l0UJr7dO%2FHDxzTWb10NviI40iww3aP                                                                                                                                                             wglLtbnA%2FVfsHMPQz21hXaCapVQ2shdG%2F%2BfhB4nujl1aoiUMErhQj5FMglGXsey3aE5M1qCbwx                                                                                                                                                             DHzenZqNZFFkmo04uT0rHFwhYklvyu23SXo0EWzOh6A4xK84S%2F%2FLYlwZ8sUFD7jtzSUkUS10XnOG                                                                                                                                                             S51c9mjXRgQgBSb1hNYsRigMkZrTCNvav2MB7mhI61atHenx5FuecuuAbS%2FBC%2FxifXQo4wnc2tIs                                                                                                                                                             pk3CvqLBWs3Ar2wCQpsPHjtww0Ib8gj45ZZkoskZUHvnZcmAues%2F4Api%2FjUwqoXikgY6wAET71dI                                                                                                                                                             2322XJIHu4zklGIS714ji%2FX%2BTbkraXpubxLeGIO90WVZGl4%2Bw1n%2Ft%2BZ62J2yeUN7zdCRjA                                                                                                                                                             qpRfhu1p63aZmnj22CaNf07qf6m3MKe3v8Ah8e2by6DZn7KmflSw90zoqLoTdRfI1Pkw8Q9ltoNPAE9X                                                                                                                                                             34qy%2BNlwjJaCFI058wPBPkzufS%2BTiBX7Inh2DPV%2FoUpou3xnuYznYinuMh7FRjjtweRSsRblS8                                                                                                                                                             UvcyY2YWDM0%2F0f33ZvAfsl3NLac%3D&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Date=202                                                                                                                                                             20414T202306Z&X-Amz-SignedHeaders=host&X-Amz-Expires=1800&X-Amz-Credential=ASIAY                                                                                                                                                             6SMA6UP472UGAWR%2F20220414%2Feu-west-3%2Fs3%2Faws4_request&X-Amz-Signature=44e5a                                                                                                                                                             23102593e9fc6d9ea1548b2f13b6d5a1c50b807a3d5243199167c48e5a0]
I (141647) AWS_OTA: Extracted parameter: [key: value]=[auth_scheme: aws.s3.presi                                                                                                                                                             gned]
I (141657) AWS_OTA: Extracted parameter [ sig-sha256-ecdsa: MEQCIDj+6THwWIhnBdbf                                                                                                                                                             tHdiZRp+Ah1H... ]
I (141667) AWS_OTA: Job document was accepted. Attempting to begin the update.
I (141677) AWS_OTA: Job parsing success: OtaJobParseErr_t=OtaJobParseErrNone, Jo                                                                                                                                                             b name=AFR_OTA-OTA_Job_1404_2222
I (141687) AWS_OTA: otaPal_GetPlatformImageState
I (141687) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (141697) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (141707) AWS_OTA: Writing to partition subtype 17 at offset 0x110000
I (142007) AWS_OTA:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (145227) AWS_OTA:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (146407) AWS_OTA: esp_ota_begin succeeded
I (146407) AWS_OTA: otaPal_GetPlatformImageState
I (146407) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (146407) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (146417) AWS_OTA: Setting OTA data interface.
I (146417) AWS_OTA: Current State=[CreatingFile], Event=[ReceivedJobDocument], New state=[CreatingFile]
I (146427) AWS_OTA: Performing TLS handshake on top of the TCP connection.
I (146437) AWS_OTA: Establishing a TLS session with xxx.amazonaws.com:443.

assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace:0x40081b8e:0x3ffdc9c00x400885ed:0x3ffdc9e0 0x4008fa1d:0x3ffdca00 0x4008948e:0x3ffdcb20 0x400df815:0x3ffdcb60 0x400d7f6f:0x3ffdcbd0 0x400d7f9c:0x3ffdcc10 0x400dc6ab:0x3ffdcc40 0x400da57f:0x3ffdcc60 0x400d9cdb:0x3ffdcc90 0x400d9d66:0x3ffdccc0 0x400da251:0x3ffdccf0 0x400d7eda:0x3ffdcd10 0x400d3164:0x3ffdcd30 0x4008bff1:0x3ffdcd50

ELF file SHA256: 6b4242d3c256d8c6

Rebooting...
ets Jun  8 2016 00:22:57
```

davidallenmann, 2022-09-30:
I get the same error with HTTP OTA.
MQTT OTA works.

robbawebba, 2022-11-21:
This bug is still active on release `202012.04-LTS` using ESP-IDF v4.4.1.

Commit `d162eb5656bfc9c324ecaf96bd72a8dc4176f53d` on the master branch fixes this issue, but has not been backported to this release branch.

Here's the patch I applied to my clone of the example that fixes the issue on the LTS release branch:

```
diff --git a/examples/ota/ota_http/main/ota_demo_core_http.c b/examples/ota/ota_http/main/ota_demo_core_http.c
index 6e36fe1..6d3144c 100644
--- a/examples/ota/ota_http/main/ota_demo_core_http.c
+++ b/examples/ota/ota_http/main/ota_demo_core_http.c
@@ -296,15 +296,20 @@ const AppVersion32_t appFirmwareVersion =
 };
 
 /**
- * @brief Static buffer for TLS Context Semaphore.
+ * @brief Static buffer for TLS Context Semaphore for MQTT connection.
  */
-static StaticSemaphore_t xTlsContextSemaphoreBuffer;
+static StaticSemaphore_t xTlsContextSemaphoreBufferMqtt;
 
 /**
  * @brief Network connection context used in this demo for MQTT connection.
  */
 static NetworkContext_t networkContextMqtt;
 
+/**
+ * @brief Static buffer for TLS Context Semaphore for HTTP connection.
+ */
+static StaticSemaphore_t xTlsContextSemaphoreBufferHttp;
+
 /**
  * @brief Network connection context used for HTTP connection.
  */
@@ -1041,7 +1046,7 @@ static int priv_connectToServerWithBackoffRetries( NetworkContext_t * pNetworkCo
     pNetworkContext->pcHostname = AWS_IOT_ENDPOINT;
     pNetworkContext->xPort = AWS_MQTT_PORT;
     pNetworkContext->pxTls = NULL;
-    pNetworkContext->xTlsContextSemaphore = xSemaphoreCreateMutexStatic(&xTlsContextSemaphoreBuffer);
+    pNetworkContext->xTlsContextSemaphore = xSemaphoreCreateMutexStatic(&xTlsContextSemaphoreBufferMqtt);
 
     pNetworkContext->disableSni = 0;
     uint16_t nextRetryBackOff;
@@ -1314,6 +1319,8 @@ static int32_t connectToS3Server( NetworkContext_t * pNetworkContext,
     /* Status returned by OpenSSL transport implementation. */
     TlsTransportStatus_t tlsStatus = TLS_TRANSPORT_SUCCESS;
 
+    pNetworkContext->xTlsContextSemaphore = xSemaphoreCreateMutexStatic(&xTlsContextSemaphoreBufferHttp);
+
     pNetworkContext->disableSni = 0;
 
     /* Initialize TLS credentials. */

```